meshcore-dev/MeshCore #7: [feature request] Add support for TX Duty cycle

recrof, 2025-01-31:
There are regions where it is required to TX using duty cycle. To be broadly-complaint, we would need to add new radio parameter with uint rage of 1-100, where 1 is lowest duty cycle and 100 is duty cycle effectively disabled.

Here is briew overview how duty cycle works: https://www.thethingsnetwork.org/docs/lorawan/duty-cycle/

ripplebiz, 2025-01-31:
There is a setting like duty cycle, but is just a simple implementation. It's called 'airtime factor' (or just AF). 
The repeater, for instance, has a CLI command: "set af {factor}"
A factor of 1, for instance will create a gap of 'dead air' equal to the time-on-air of the packet it just sent.
A factor of 2, creates a gap equal to 2 times time-on-air, etc.

You can set it to any floating point number. A zero means no gaps, ie. duty cycle 100.

closed 2025-02-01 by ripplebiz

recrof, 2025-02-16:
can we have this option for companion firmware aswell please?

chixxi, 2025-02-16:
Is this implemented in the T-Deck firmware too? And if yes how do I set it? Should probably be an option in Settins=>Networks