meshcore-dev/MeshCore #9: Command line tool to interact with nodes in BLE

fdlamotte, 2025-02-02:
Thanks for providing the BLE companion code, it will help me a lot 

I did some tests this we and it resulted in writing some python code to interact. I managed to send messages, get the contacts and other things.

The code is here : https://github.com/fdlamotte/mc-cli 

Maybe someone has already done it but I could not find anything ... for the moment the BLE address of the node is hardcoded, and send command works in python but not from the shell (should send the pubkey as hex ;)) but sendto uses the adv_name of the contact so it works ok

Hope this helps ;)

regards

ripplebiz, 2025-02-04:
Good to hear the BLE stuff works. I haven't got anything to test it against yet, so have been kind of flying blind with it :-)

fdlamotte, 2025-02-04:
Cool,

It happens I have a linux phone (p3a running postmarketos) and the usb otg is not totally functionnal (can't power the device yet). So the only way I can talk to nodes is through wifi or ble.

I'll continue improving the tool and will share the issues I have

About retreiving received messages, I've just seen that it was not implemented yet ;)

ripplebiz, 2025-02-04:
Ah, yes. You've reminded about that 'TODO' still sitting there:
   addToOfflineQueue(), getFromOfflineQueue()

I'll get onto that in a sec.

recrof, 2025-02-04:
> Ah, yes. You've reminded about that 'TODO' still sitting there: addToOfflineQueue(), getFromOfflineQueue()
> 
> I'll get onto that in a sec.

Oh, I was surprised why my desktop app doesn't get the actual message when I get the push frame! let me know in discord when this will be done please

ripplebiz, 2025-02-04:
OK, that should be done now.

fdlamotte, 2025-02-04:
ok, thanks, managed to get msgs ;)

fdlamotte, 2025-02-04:
I've reached the state I wanted to go ... Did some documentation on the usage, I'll probably refactor things a bit and will try to keep in sync with the tree

This tool serves me well, I hope it can help the community ;)

ripplebiz, 2025-02-04:
Just found a nasty bug in the companion radio firmware. Fixed now.

fdlamotte, 2025-02-06:
thanks for no_more_messages response ;)

recrof, 2025-03-12:
@fdlamotte can we close this?

fdlamotte, 2025-03-12:
yes, i think so ... closing it myself ;)

closed 2025-03-12 by fdlamotte