meshcore-dev/MeshCore #13: Support for T1000-E

fdlamotte, 2025-02-04:
I have one of these, still loaded with meshtastic firmware (and I keep another meshtastic node to be able to use it because I have nobody else in reach) but I've not used these since I've started playing with meshcore.

I was wondering how difficult/long it would be to get it working on meshcore (with the ble companion firmware)

From what I read : 
 - it is based on nrf52 (there are already boards with it but no ble yet if I recall)
 - it uses an LR1110 for lora (don't know how far it is from sx126x) 

One of the issues I have is with the flashing, which is done using a virtual drive ... I don't like it much because I don't really feel in control.

I have another concern about powering the thing off, because the device does not have an hw switch (and don't want to open mine). Either I'll have to keep it powered on, or I'll manage sleep/off states with the button.

Has anybody considered using this device with meshcore ?

Dylanliacc, 2025-02-05:
I don't think it's out of reach, RAK4631 is also NRF52, and meshcore also using radiolib,we will figure it out.

fdlamotte, 2025-02-05:
You're right, I don't have experiece yet with nrf52's and t1000-e product page focuses a lot on bricking the device so that's why I'm a little cautious (also the fact that it is the device I had the hardest time trying to figure out how to flash it)

I'll experiment with it and keep you informed ;)

fdlamotte, 2025-02-06:
Hi, got it working out of tree on the terminal chat

https://github.com/fdlamotte/T1000eTerminalChat

I'll try to make a PR, but I don't have the exact same definitions in my nrf52 environment as for the rak boards ... need to sort things out ;) but that is a start

ripplebiz, 2025-02-06:
Ah, cool. Well done.
I'm still not sure what 'the rules' are when incorporating source files from the Meshtastic repo? ie. with the GPL3 license. For now, your fork might need to stay in the separate repo, until I can find out.

fdlamotte, 2025-02-06:
No problem, that is a valid concern ...

I have used the bsp from meshtastic, (from boards and variants directory). Variants file have last copyright from Adafruit in a LGPL licence. Normally these should be given by pio when configuring the board (I think there is a similar issue with rak boards)

I was also using a framework-arduino-adafruit from ms as platform, but the original one seems to work ... I did that while investigating why ms code compiled while mine did not (with issues with bluefruit include in apple homekit ...)

all the code in src derive from your work (I've adapted your helpers for board and radio), I think this can be safely incorporated in your tree. I can make a PR with only that, to pave the way for a broader incorporation ?

And for the current issue with the t1000, it seems that the radio goes to sleep and doesn't wake after some time ... not too bad ;)

Dylanliacc, 2025-02-07:
> You're right, I don't have experiece yet with nrf52's and t1000-e product page focuses a lot on bricking the device so that's why I'm a little cautious (also the fact that it is the device I had the hardest time trying to figure out how to flash it)
> 
> I'll experiment with it and keep you informed ;)
> 
> edit : just tried to build and upload meshtastic fw using vscode and it went smooth ... don't know why they make things so complicated in the docs ...

don't worry about that ,if don't break bootloader ,it won't be really easy to brick ,otherwise if really brick u can dm me for a  new one

fdlamotte, 2025-02-10:
Got some time to work again on the T1000e using terminal chat, still facing an issue where the t1000 stops receiving data ...

When sending consecutive messages, the t1000e receives the first and acks it but then does not receives anything (no msg and no adv) until it sends a message ...

If I remove the ack part in BaseChatMesh::onPeerDataRecv then everything works, so I suspect there is something wrong with the ack but I have not yet found what could be different there from a std msg ... I'm continuing but if someone has an idea ;)

fdlamotte, 2025-02-11:
I have some new insight ...

If I force the payload lenght on the T1000 to 20 there are no more blocking ... would there be a minimum payload size ?

Also if I send a packet to the t1000, with a payload > 20 it blocks (the t1000 does not receive it). But the t1000 can send packets with payloads over 20

More interesting if I send a packet with a payload over 20 from the t1000, and send back the packet to the t1000, it works ... so maybe I have some sort of buffer allocation issue

fdlamotte, 2025-02-11:
went to read meshtastic code to get some inspiration on how they handle the LR11x0 and found that :

`lora.setPreambleLength(preambleLength); // Solve RX ack fail after direct message sent.  Not sure why this is needed.`

It might be needed because if I do this call, things do work for meshcore (both issues with ack and longer payloads !)

I've added this at the begining of RadioLibWrapper::recvRaw

`   _radio->setPreambleLength(8);`

I'll find a better place for it ;)

fdlamotte, 2025-02-14:
position and time synced with the internal gps ;)
I'm ready to go skiing ...

closed 2025-02-28 by fdlamotte

chiefMarlin, 2025-03-21:
Works perfectly so far!