meshcore-dev/MeshCore #21: What makes this project different from Reticulum?

LinuxinaBit, 2025-02-14:
The README is very vague, and I can only see that it is somehow ‘simpler’ than Reticulum with a focus on embedded devices while still scaling well.

When and why would someone want to use this instead of Reticulum? What makes it special?
I would like *details*. Routing specifics, design choices, *everything*.

recrof, 2025-02-14:
From what I observed, it's much more focused on IM. it doesn't want to have full network stack. MeshCore is primarily designed to work on MCUs, not on computers.

liamcottle, 2025-02-14:
From my experience so far, this is how you use it:

- You flash the [companion firmware](https://github.com/ripplebiz/MeshCore/blob/main/examples/companion_radio/main.cpp) to a [supported board](https://github.com/ripplebiz/MeshCore/blob/main/platformio.ini)
- You connect to it with one of the existing clients over BLE or Serial
  - https://meshcore.liamcottle.net
  - https://meshcore.dualsoul.net
  - https://github.com/fdlamotte/mc-cli
- You set a display name
- You configure your LoRa parameters
- You send an _Advert_, which is similar to an _Announce_ in Reticulum terms
- You can send and receive direct messages with other users on the network
- You can send and receive broadcast/flood messages with other users on an encrypted _Channel_
- There are some other things such as Room Servers, that I have not played around with yet, so won't be covered here.

I've only been playing around with it for the last few days, but have written a new web client and deep dived the firmware and packets. Based on this my understanding is the following;

- The companion firmware generates an ed25519 key pair on boot and persists it to the file system
- When you send an _Advert_ it includes your public key, the current timestamp and some optional application data
  - In the case of the Companion Firmware, this is your display name and optionally a configurable latitude and longitude
- Your _Advert_ can either be flood routed, or zero-hopped
- You can flash the Repeater Firmware to another board, which is very simple. If it receives a packet marked for flood routing, it adds the first byte of it's own public key to a path list on the packet, and then rebroadcasts it.
- The next repeater will hear the rebroadcasted flood packet, and again add itself to the packet path list, and rebroadcast it.
- Repeating is allowed up to 64 hops.
- When another device running the Companion Firmware receives the _Advert_ it can add it to its internal contacts list, similar to how we show Reticulum _Announces_ in MeshChat as they come in.
- Additionally, since the packet received contains the path it took to get there, the node that received it can then send that path back to the original sender in another flood packet, which tells the original packet sender how it's own packet got to the destination.
- From then onward, the original node can use the same direct path for as long as it works, or until the user decides to _forget_ the path, in which it will revert back to flood routing.
- This can be done both ways, each user sending a flood in each direction to establish the path.
- Once a path is established, a packet can be sent in direct route mode, which includes the path list (first byte of each repeaters public key), and when it is broadcasted, only the repeaters who match that first byte will repeat the packet
- If you know what you're doing, you can manually create a path, with specific repeater ids you want to go through

Lots of info to digest, I hope it's helpful. Having deployed Meshtastic networks across hundreds of kilometres, and being limited to 7 hops, it's quite nice to finally have a setup that allows for up to 64 hops with very very small packet sizes, and from my testing it's super fast since the repeater doesn't try to do anything else other than repeat :)

All 3 of these projects, Meshtastic, Reticulum and MeshCore all have their strengths and weaknesses, and they also have their own goals which differ greatly from each other.

From what I've learned and experienced with it so far, it seems like a very good choice for simple and fast mesh based messaging over LoRa.

I'd also like to note that things have been abstracted in a way that make it relatively easy to make a custom, super minimal firmware as the Companion Firmware, Room Server Firmware and Repeater Firmware code are all just separate `main.cpp` example files that use the open source Mesh classes in this repo to do the fun stuff.

ripplebiz, 2025-02-14:
Thanks for the descriptions Liam.

I'd just add that possibly the biggest difference to Reticulum is that MeshCore is LoRa-first, and aimed at micro-controllers. It has a very small footprint, can run on extremely modest hardware, and the packets formats are very frugal/tight. So, it's designed for performance, in this very resource constrained environment.

It's also very extensible. There are some basic packet types, like adverts, acks, etc, but a lot of the specifics on _content_ and structures that can go over this base protocol are very flexible and open ended. Also, as Liam has pointed out there isn't 'The firmware', ie as one monolithic blob. The core logic is in library form and easily customised for different nodes/roles.

And MeshCore is also _very_ privacy-centric. Not a lot has been mentioned about this, but there are NO (complete) source/dest ID's in packets (only partial hashes). No packet IDs. Source/dest pairs can remain completely anonymous, and actually don't have to advertise/announce for the routing to work (like in Reticulum). They can 'know' each other by other means, and still get end-to-end connectivity that is completely private/encrypted.

LinuxinaBit, 2025-02-14:
Some of that is pretty interesting.

The LoRa focus and microcontroller focus is a tradeoff that I completely understand - one of the biggest features of Reticulum is that it doesn’t focus on *specific* physical layers or devices like that; also a tradeoff - what MeshCore’s focus gains is the ability to have better, more seamless LoRa support at first, but it loses a big amount of versatility, at least initially.

If I am understanding correctly, MeshCore flood routes packets most of the time, but can form Reticulum Link Destination-style connections using an overlaid system of adverts?
That’s… interesting.

That almost certainly still has a measurable negative effect on efficiency, but it makes a kind of broadcast/pseudo-multicast easier. I think it’s *possible* to make it scale to an extent, but there will almost certainly be a point where that becomes much harder.

Still, I do think this may have some potential.
I hope to see you around! 👋

LinuxinaBit, 2025-02-14:
> From what I observed, it's much more focused on IM. it doesn't want to have full network stack. MeshCore is primarily designed to work on MCUs, not on computers.

@recrof Unless you’re confusing Reticulum and Meshtastic, I don’t see how you could believe that if you understand Reticulum to a reasonable degree.

Do check out the Reticulum explainer I wrote a while back if you want to learn more: https://linuxinabit.codeberg.page/blog/reticulum/

recrof, 2025-02-14:
> > From what I observed, it's much more focused on IM. it doesn't want to have full network stack. MeshCore is primarily designed to work on MCUs, not on computers.
> 
> [@recrof](https://github.com/recrof) Unless you’re confusing Reticulum and Meshtastic, I don’t see how you could believe that if you understand Reticulum to even a reasonable degree.
> 
> Do check out the Reticulum explainer I wrote a while back if you want to learn more: https://linuxinabit.codeberg.page/blog/reticulum/

I'm not sure if you understood my message, and I'm pretty confused you even mentioned Meshtastic. I'm familiar with Reticulum and MeshCore. 

Let me clarify: MeshCore was not designed to transfer arbitrary network traffic. MeshCore is LoRa messaging system. Reticulum does not (yet) fully run on microcontrollers(microreticulum was not complete last time I looked).

closed 2025-02-21 by ripplebiz