meshcore-dev/MeshCore #24: Repeater debug: No way to identify links

Andrew-a-g, 2025-02-15:
The output below shows RSSI / SNR etc which is great but to which node there is no way of telling.
```
PACKET: recv, len=6 (type=3, route=D, payload_len=4) SNR=7 RSSI=-59 score=1000
PACKET: recv, len=38 (type=2, route=D, payload_len=36) SNR=8 RSSI=-58 score=1000
PACKET: recv, len=6 (type=3, route=D, payload_len=4) SNR=6 RSSI=-53 score=1000
PACKET: recv, len=113 (type=4, route=F, payload_len=109) SNR=6 RSSI=-96 score=1000
DEBUG: Dispatcher::checkRecv(), score delay below threshold (0)
DEBUG: Mesh::onRecvPacket(): valid advertisement received!
PACKET: send, len=114 (type=4, route=F, payload_len=109)
PACKET: recv, len=114 (type=4, route=F, payload_len=109) SNR=8 RSSI=-57 score=1000
DEBUG: Dispatcher::checkRecv(), score delay below threshold (0)
```
Is it possible to add a unique identifier to each line with contact identifier (e.g. even a couple of unique bits) to track signal changes over time when tweaking antenna setups?

liamcottle, 2025-02-15:
Would love to see the SNR exposed to the client apps.

It would be fantastic to be able to get signal stats on a _per message/packet basis_ as they come in, as well as when a contact/repeater sends an advert, so I'd be able to show their signal metrics in the contacts list.

Here's a couple of screenshots of some info I show for announces and messages received in my MeshChat application when received over LoRa (in a Reticulum network).

[image]

[image]

Andrew-a-g, 2025-02-15:
> Would love to see the SNR exposed to the client apps.
> 
> It would be fantastic to be able to get signal stats on a _per message/packet basis_ as they come in, as well as when a contact/repeater sends an advert, so I'd be able to show their signal metrics in the contacts list.
> 

I like the idea of this but the problem is the stats on the client are just the last hop to the nearest repeater..  its actually the signal strength between the repeaters that is most important. To better build networks we need better visibility and logging of RSSI / SNR per direct contact on the repeater

446564, 2025-06-19:
Do we have this now with the neighbors list?