meshcore-dev/MeshCore #28: set initial default frequency, BW, SF values per region

LitBomb, 2025-02-16:
As more users coming on board, it would be great to set the initial frequency for various regions. Andy on discord suggested that we could have a set region command.  If implemented, this feature will help standardize the initial default frequency per region, so new users with new radios running meshcore will get set to a frequency that will most likely have other radios on and start communicating as we grow the user base around the world.

For example, use these values for USA:
  -D LORA_FREQ=910.525
  -D LORA_BW=250
  -D LORA_SF=10

910.525 MHz should be far enough away from popular frequencies used by other mesh systems.  This would just be the initial defaults.  Users should still be able to set to different settings if they so choose.  

Thanks!

chixxi, 2025-02-19:
I think it would be worthwhile considering to also make "set af {air-time-factor}" and "set tx [power dBm]" part of these regional presets as there can be different legal requirements in different regions for these settings.

shortwavesurfer2009, 2025-03-14:
This would be a good idea. The default that's currently set is actually completely illegal in the United States, for example.

liamcottle, 2025-03-14:
My preference would be to have the firmware default to `LORA_FREQ=0`, in which case transmit is disabled when initially flashed. It would be up to the user to set the frequency via serial, or the apps to display a recommend frequency/preset. Otherwise, as noted, when the boards are flashed with a default, they will transmit on frequencies they shouldn't when they first boot up.

prunes-git, 2025-03-22:
I would love to see the radio disabled and a list of regional default frequencies, there is already a decent list on the discord for defaults, I have seen someone here trying to use the current default here which is out of our ISM band.

446564, 2025-06-19:
We have the default for different countries now so we could switch to fresh firmware having freq set to 0 and this is done.

chixxi, 2025-06-20:
@446564 I think the Air-Time-Factor should also be included in the presets before we can consider this one done. It's an important part of using MeshCore nodes while complying with the legal restriction. Or at least we should create a follow up issue for that if we close this one.