meshcore-dev/MeshCore #31: ATAK plugin

vygidor, 2025-02-19:
Add the functionality to integrate MeshCore with ATAK on Android devices.
Note: the functionality must support devices both without GPS module and with GPS module.

example: https://meshtastic.org/docs/software/integrations/integrations-atak-plugin/

liamcottle, 2025-02-20:
@recrof I've never used ATAK, so probably wouldn't add support myself.

W1BTR, 2025-07-08:
+3 on this. Love ATAK and the idea of a cheap mesh service that can do this, but Meshtastic has shown itself too unreliable especially with many devices. Meshcore "Just Works" to a much higher degree.

liamcottle, 2025-09-16:
Closing out for now. I don't think we plan to, or need to add anything in the MeshCore project itself for ATAK.
If this is something you want to use over MeshCore, you could write your own wrapper program that exposes some interface for ATAK to be able to use the custom raw data packet type that you can send data through.

closed 2025-09-16 by liamcottle