meshcore-dev/MeshCore #35: Serial Communication Changed?

Jobs2k, 2025-02-21:
What changed since commit cf63ec9 that has prevented/significantly changed accessing a device through serial?

I have been happily playing with some code to interact with some Heltec V3s I got specifically for Meshcore. Now I can't use anything outside of the web apps. The device just hangs on connection until a complete power-cycle.
I assume I now have to tell the device "I'm here, let's do this" but I have no idea how I would go about doing that.

recrof, 2025-02-21:
try https://github.com/fdlamotte/mc-cli - if this works, something is wonky in your code, because 3 other implementations work(my MeshCoreClient, liam's web client and this python cli).

Jobs2k, 2025-02-22:
I can completely accept my code being entirely wonky! I'm not C or C++ able, let alone competent.
I'm just playing in Python and was using PySerial to essentially emulate user input and handle responses.

Just casting my eyes over the code you linked suggests that I'm not meant to be doing that and will have to dig even deeper to have my fun. :(

Jobs2k, 2025-02-22:
I have given this quite a bit of thought and figured I should go back to the way I began this journey.
Here are the steps I took:

```
- Heltec unplugged - unpowered, no battery.
- Open web app in browser. 
- Heltec connected to USB.
- Click 'CONNECT SERIAL'.
- Select device, click 'Connect'
- Text claims: "Connected to [node name] via USB Serial".
- Click 'DISCONNECT'.
- Close browser.
- Remove Heltec from USB.
- Open PyCharm, activate 'Serial Port Manager - by JetBrains'.
- At this point there is a tangent detailed below - I don't want to break the list's flow.
- Connect Heltec to USB.
- Select new (only) device in 'Serial Port Manager' and click 'Connect'.
- No welcome message is seen, no user input is shown in the panel, return key provides no response.
- Click 'Disconnect'.
- Close PyCharm.
- Remove Heltec from USB.
- Open CuteCom.
- Connect Heltek to USB.
- Select available USB device and click 'Open'.
- Receive this message:

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x28 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0

- Type "Help" in the input field and hit 'return'.
- No response displayed from device.
- Click 'Close'
- Close CuteCom.
- Remove Heltek from USB.
```

Tangent:
All successful communications have been through ttyAM[n] , however the connections are now going through ttyUSB[n].
I got a list of available serial devices with with PySerial and it claims it is a filesystem object that is rejected as a Serial object.

So something does seem to be different, as all this was done without my code involved.

What direction can I take now? What type of code is going to nudge this in a productive direction?
As I used the web flasher available at ['meshcore.co.uk'](meshcore.co.uk/apps.html) is there any chance it provided the wrong software?
I haven't found any nodes (that aren't mine) in my area so I can't confirm any communication functionality. All my devices are on the same software version.

recrof, 2025-03-13:
@Jobs2k not sure if I have been any help. is this issue still relevant? we can't do much on our end. Close the issue as you see fit.

Jobs2k, 2025-04-09:
Nothing I used to do allows previously available serial commands. I've eliminated any chance of it being my code by updating to many versions (currently 1.4.3) and directly accessing the device with serial software that responds exactly as expected with a repeater.
Something has to have changed as no node will respond to 'raw' serial text/commands as they used to, and my repeater (also on version 1.4.3) currently does.
The only clue I have is that meshcore.co.uk states in the flasher that the console only works for repeaters and room servers. There are many corresponding commands there.

I'll have to assume that I need to do a lot more than just send text through serial and I don't have the knowledge for that.

closed 2025-08-06 by recrof