meshcore-dev/MeshCore #36: Hardware Request for Lilygo T Beam to be added

Hesit8or, 2025-02-22:
Hardware Request for Lilygo T Beam to be added, Just simply due to Popularity of the device

ripplebiz, 2025-02-22:
Can you provide a link of the _exact_ model? There are a number of T-Beams, unfortunately, with different GPIO's and power management chips, etc.

Hesit8or, 2025-02-22:
TBeam Supreme please

k1e2n3t4, 2025-03-04:
standard tbeam off of amazon  915 Mhz  NOT supreme version.

k1e2n3t4, 2025-03-04:
https://www.amazon.com/LILYGO-Development-LoRaWAN-Wireless-Battery/dp/B0CR18RY5N/ref=sr_1_5?dib=eyJ2IjoiMSJ9.KxfjlmyL0WlhZl1ztLjpQpj6bvKl-ezer3-y_pjAMTFJD2IpPOvYjOF7r8dlEXpZNya9b2mX1v-xWATOuySigUhJ4rjZ2oMZ14j1lpmWZQYw4GsaHbhjVV4xhHNEt9FznP65prtCU59BC3w1l4v7BqujcjDnjyma6uJez6iy7MVhk0QN_dYDrWh3-8HPoLGvpMtCt4Jsg6qwlV9M-mdAHLypkaBYd_n5RJkJxl2TTuE.hTLUo97UaSRH2aZDmq6vs6xYFRtsnrjOkj81D2il7UE&dib_tag=se&keywords=tbeam&qid=1741117474&sr=8-5

Atorcha, 2025-03-11:
[image]

Hardware: SX1278 (433 MHz) SX1276 (868/915 MHz)

T-Beam "meshtastic" version.

https://www.amazon.de/dp/B0B454TB6S/ref=pe_27091401_487027711_TE_SCE_dp_i1?th=1

jameshall9327, 2025-03-13:
i just did a pull request for the support for the lilygo tbeam v1.2 with the sx1276

jameshall9327, 2025-03-13:
> standard tbeam off of amazon 915 Mhz NOT supreme version.

What version i added support for the v1.2 with the sx1262

Johnex, 2025-03-15:
I have 2 T-Beams with SX1262 (868/915mhz) together with the Neo-M8N. At the time i bought it, this was considered the best model. The lower model chip actually performs better than the higher model chip from looking at reviews, and the M8N GPS i have on all my custom drones, which is also the best GPS they have on the T-Beam.

I currently have them on Meshtastic but i would love to move them over to MeshCore, when support is added. Thanks so much for the amazing project btw :)

jameshall9327, 2025-03-15:
> ) do you have a way to receive a bin file?

jameshall9327, 2025-03-15:
Do you have a way to recieve a bin file?

> I have 2 T-Beams with SX1262 (868/915mhz) together with the Neo-M8N. At the time i bought it, this was considered the best model. The lower model chip actually performs better than the higher model [...]

Johnex, 2025-03-15:
@jameshall9327 Are you on the meshcore discord? I just joined there now.

jameshall9327, 2025-03-15:
> .

Yes jmhmesh is my user name

Johnex, 2025-03-15:
@Hesit8or @jameshall9327 Saw that the "Supreme" version is also the SX1262, just different GPS. The power stuff might be different though, the layout is completely different, they are using a daughterboard.

jameshall9327, 2025-03-31:
I will be adding support for the SX1262 tbeam board soon

rusty-labs, 2025-04-23:
I have created pull request for the device https://github.com/ripplebiz/MeshCore/pull/218

seagull9000, 2025-04-30:
I have some questions re TBeam V1.2

I have flashed with the v1.5.1 companion radio ble firmware.
1. Should the TBeam v1.2 show battery/plug status?  I just get "Battery 0%" in the App.
2. Does the user button work?  I see there are a number of messages but can't see a way to scroll through them.

recrof, 2025-04-30:
2. user button just dismisses the messages, scrolling through them is not implemented

446564, 2025-06-20:
Is this complete now?