meshcore-dev/MeshCore #37: [Enhancement] Non Lora Meshing (e.g. via ESPNow, TCP/UDP via WiFi)

Noki, 2025-02-22:
I would really love to see additional options for letting devices mesh in addition to using Lora e.g. via ESPNow, or TCP/UDP via WiFi. Here are some use cases:

- Make it easy to bridge 433 Lora and 868 Lora networks
- Reduce Lora airtime and avoid duty cycle limits for small distances (e.g. roof repeater / indoor node) maybe even making Lora optional for an indoor node
- Make it easy to bridge long distances via existing infastructure (e.g. via a WiFi mesh network or the internet in general)

I have no idea if these are viable options, but I definitely wanted to share them here and have them documented.

ripplebiz, 2025-03-04:
Yes, I have been musing over adding these capabilities. Definitely do-able.
I definitely want to explore stuff like this.

yNosGR, 2025-03-08:
Having a KISS/TNC interface would be nice as well. For example, being able to bridge/mesh over a serial TNC of over a tcp port like FreeDV-TNC2.

kurtrenauer, 2025-03-15:
MQTT-Bridge would be welcome very much as well, for my use cases and existing infrastructur.

Atorcha, 2025-03-17:
MQTT usefull

sevenbitbyte, 2025-03-18:
I'd like to see Serial and possibly CANBus on the list.

xpinguinx, 2025-04-07:
Integrating MQTT protocol would be awesome!

ZilvinasJonaitis, 2025-04-22:
I wish I could connect two LoRa nodes by wire (Serial). This could be especially useful for outdoor repeaters comprising of two repeater nodes in one box. One repeater dedicated to local communications within a range of omnidirectional antenna, another repeater node dedicated for long-haul link to remote repeater(s) via directional antenna.
Of course, there should be some interference management protocol implemented as both repeaters will operate on the same LoRa frequency. Some minimal serial link delay with buffering could a solution.

jbrazio, 2025-06-07:
Just my two cents, I would really hope Meshcore don't go into the direction of non RF bridges just like the other we all know.

TyIsI, 2025-06-13:
@jbrazio Honestly, I would argue that bridging to/over different media is not bad, as long as it doesn't become a default backhaul over the internet solution. Creating seamless extension/integration over distance/frequencies is what can really make it resilient.

Or even running local links on different frequencies than backhaul links.

kryptonut, 2025-06-17:
> [@jbrazio](https://github.com/jbrazio) Honestly, I would argue that bridging to/over different media is not bad, as long as it doesn't become a default backhaul over the internet solution. Creating [...]

Agree 100%, would be extremely limiting to not be able to have some kind of backhaul to extend the range of a Meshcore network.

recrof, 2026-02-07:
we will not allow tcp/ip bridging as it encourages users to use internet as a bridge between the meshes - it causes unwanted traffic that can clog meshes pretty quickly.

KenSamson, 2026-02-10:
> we will not allow tcp/ip bridging as it encourages users to use internet as a bridge between the meshes - it causes unwanted traffic that can clog meshes pretty quickly.

I can see how that is a legitimate concern.   So maybe the right way to allow 2.4Gig to interop with 915Mhz in the US would be through USB bridging only?   I don't know if the devices can be a usb host.. so maybe a pi between them?   Ideas anyway.  Thanks for the reply.. it makes sense given the history.   I'm hoping if regions and repeaters catch on well enough that traffic can be contained.   I'm sure there will continue to be evolution.

christheradioguy, 2026-02-12:
> > we will not allow tcp/ip bridging as it encourages users to use internet as a bridge between the meshes - it causes unwanted traffic that can clog meshes pretty quickly.
> 
> As long as CLI is available, anyone can write a bridge. As long as the protocol is open, anyone can write an interface (especially with the modern AI tools, however messy the result will be).
> 
> mqtt is more than enough. For one, I'd like to bridge my silent Room server with my house repeater over wifi or ethernet (or, even better, on my RPi/Docker based room server, no direct LoRa access at all). No need to have TWO LoRa nodes if all I need is to chat on the network, right? It's actually three, right -- the third one being my companion terminal.
> 
> If three nodes are considered normal for the casual home use, I'd hazard an opinion that allowing linking over IP (or other radio) would be actually beneficial -- instead of adding 10 more routers on the balconies and roofs, city users could simply use IP to connect to their favourite local repeater positioned on a tall radio tower.
> 
> This is what we do in the ham world -- no one in their right mind deploys their "home repeater" using their spare radio's cross-band repeater function only to talk from the living room with their handheld - instead we deploy repeaters in the strategic, high places.
> 
> As long as the protocol is open, anyone can add IP linking as an extension, and a few will. I will use the mqtt route myself to forward my house room server chatter to my phone whilst I'm away.
> 
> Blanket refusal will only contribute to the fragmentation of the tooling and encourage protocol extensions that you wouldn't want to write/add yourself and maybe a path into adopting LF-FHSS would be a sane way to increase the capacity and resiliency of Meshcore, especially as the WiFi HaLow (801.11AH) has a chance of getting widely adopted.
> 
> Of course this is strictly __your__ network protocol as I understand so no means no, i'm just adding my perspective.

Does MC currently support MQTT over TCP/IP?  That would actually solve my problem of retrieving messages locally over IP

pramodhrachuri, 2026-02-23:
I understand the maintainers' fear that people will just use the internet instead of building standalone bridges, and adding the internet will easily overpopulate things. Imagine 10 local meshes connected over the internet, and suddenly one node starts flooding. Might lead to significant resource wastage.

But at the same time, the restriction limits scaling up in the initial days. Using existing infra will drastically improve coverage and maybe adoption. vrybdpkt's case is a great example. We can always impose restrictions and penalties on TCP/IP bridges. For example, a much lower allowed hop count, or simulated extra hops+delay for making the routing algo avoid these bridges when they can.

@vrybdpkt great work! Are you still planning to make a pull request and give it a try, or at least create a patch when you're done, so we can use your work?

vrybdpkt, 2026-02-23:
I understand the concern however for it to work there has to be a coordinating other side unless someone stands up a public broker that is unmoderated. For that case I've implemented banning though I haven't tested that yet. I can also see other use cases like emergencies where you would want to be able to connect to something that otherwise isn't available. 

@pramodhrachuri 
I'll open a pull if there is real interest however I'm just leaving it in my fork for now. It's up to date.

smellyspice, 2026-08-30:
Hi @controlrepo and @christheradioguy — I put together a fuller write-up of ESP-NOW meshing (companion, room server, or repeater with no second LoRa radio) plus an IP bridge addressing the flooding/scope concern raised in this thread, as its own Discussion rather than repeating it all here. You both described almost exactly these use cases a while back, so thought you'd want to see it:

https://github.com/meshcore-dev/MeshCore/discussions/3327