meshcore-dev/MeshCore #41: Sensor Support

mepholic, 2025-02-23:
Hello, I'm extremely interested in deploying a MeshCore network, but I would like to see an efficient mechanism built into the protocol for reporting sensor measurements.

I see 2 other issues which cite the general desire for this to exist in the protocol, and that would be this particular post [here](https://github.com/ripplebiz/MeshCore/issues/11#issuecomment-2638453431), as well as this very good thread [here](https://github.com/ripplebiz/MeshCore/issues/1).

I'm hoping to create this issue as a general tracking ticket for sensor integration into MeshCore.

I specifically want the ability to get stats from Voltage and Current sensing I2C devices like the INA219 and INA3221, so that I can measure and report battery pack stats for solar nodes.

Please let me know if there is any way I can assist in the testing or development of this feature!

greghall-au, 2025-03-09:
Telemetry will add immensely to the mesh. 
Some of us are trialling ways to send (& display) environmental conditions, especially extremes that inform public safety. 
Two examples are extreme rainfall and water depth (e.g: over causeways). 
https://www.floodnet.nyc/ and https://coast-lab.org/waterLevel/ are doing great work on this, BUT they both use the mobile phone network to communicate. Not very resilient!
- 
Meshtastic grew an eclectic collection of sensor built-ins, as enthusiasts coded them - but nothing like a standard...
We've trialled feeding our rain and flood obs into Meshtastic as serial, after pre-processing by a second micro. OK, but a bit clumsy.
-
Data sentences will need to be very compact, to avoid cluttering the airwaves. APRS might have hints to offer here, especially the Rain: 1Hr, Since Midnight, 24Hr but the Imperial units - Yeccch! Surely there's room for XML or JSON, if we use it efficiently.
-
Displaying the data effectively is another task, we can't rely on continuously-available servers to aggregate and generate charts, etc. This needs to be done in the client's device, probably by the Mesh app, writing out it's data to be available to other processes. The Room Servers might have a role here though.

(Just thinking aloud;)
73
Greg
vk2gth
-

fdlamotte, 2025-03-13:
I've tested a client/server approach based on the repeater cli

https://github.com/fdlamotte/SimpleNode/tree/main

Also light_sleep is enabled in this code