meshcore-dev/MeshCore #45: Heltec v3 not retaining epoch time on disconnection from Console

mjvolante, 2025-02-24:
Trying to set Heltec v3 as Repeater.

Heltec v3 Ver3.1 (with battery) reverts back to May 2024 on disconnection form either Meshcore Console or https://googlechromelabs.github.io/serial-terminal/

Tested two different Heltec v3's (flash with erase), two different batteries. (Batteries charged)
Port connect request reads: CP2102 USB to UART Bridge Controller COM3 - paired
Browser: Chrome

[image]
[image]
[image]
[image]

mjvolante, 2025-02-24:
Please note the above screenshots are when the USB power is permanent with no cable disconnection from PC and no reliance on battery power.

liamcottle, 2025-02-25:
Ok, I've confirmed this clock issue with Heltec v3 not sticking when disconnecting from serial port is a windows issue. It will be an issue with all boards, and not just this one.

Do note, for both of these tests, the usb was never unplugged, when I say disconnect, I mean I closed the software serial port by clicking disconnect in the web browser.

**macOS**

With both the flasher.meshcore.co.uk serial console, and the chrome web serial console (used above), time is set, clock shows correct time, and board does not reboot when disconnecting from the board in software. Upon reconnecting, the correct time is provided by the board.

**Windows**

Both the serial consoles listed above set the time, but when disconnecting and reconnecting to the serial port again, the board reboots itself. The reboot happens on disconnect, and the reboot happens on reconnect.

Due to the board rebooting, the time is lost.

This issue with serial ports is a common thing on Windows, lot of issues in the Arduino forums about it. Caused by RTS/DTR being set upon board opening. As far as I know, it's not possible to configure this in the javascript clients either, because Windows sets the RTS/DTR _before_ the client apps have a chance to say not to.

**Possible Solution**

One idea I had, was to store the timestamp in a file on flash when the user runs the `time <timestamp>` command to set the clock. Upon board reboot, we could restore the last known timestamp.

This would allow the board to have a near enough timestamp after the board reboots after configuration, however it does not solve the case where the board may reboot several days later (for whatever reason), as the old timestamp from flash will now be outdated. It doesn't make sense to write the timestamp to flash frequently either...

Ideally, we will have remote clock sync over the network, I proposed a custom MTP clock server (Mesh Time Protocol), but I won't list all of that here right now.

ripplebiz, 2025-02-25:
OK, with the latest commit, you can now wire up a DS3231 RTC module to the Heltec, and it should be automatically detected. 
(The SDA/SCL pins for the Heltec v3 are actually on the back, and are quite tiny, unfortunately.)

closed 2025-08-31 by liamcottle