meshcore-dev/MeshCore #51: Feature Request - Find Path Button

SubnetConflict, 2025-02-26:
Feature Request - Find Path Button

Local testing involves a lot of messaging pt pt pt pt or random characters in chat till a path has been found.
Can we have an additional button added for 'Find Path' then it informs us if a suitable path has been found please?

Think Longer term this will help people not get irritated with loads of failed messages before a path is found.

Thanks

Marc

banerian, 2025-09-03:
app.meshcore.nz   doesn't let you view paths for a given message, or if it does, it's not at all obvious, unlike the mobile app which does do it via a long press on the message.

should this be a feature request for the app-to-usb-serial/ble ?

banerian, 2025-09-04:
> > app.meshcore.nz doesn't let you view paths for a given message, or if it does, it's not at all obvious, unlike the mobile app which does do it via a long press on the message.
> > should this be a feature request for the app-to-usb-serial/ble ?
> 
> Did you try long clicking with the mouse?

yes.  long click, center click, right click.  no change

closed 2025-09-16 by liamcottle