meshcore-dev/MeshCore #53: Unable to flash Lilygo T3S3 v 1.2 1276

fenixstarlord, 2025-02-28:
When attempting to flash a T3S3 1276 the web flasher reports that the flash is complete, but uppon restart the MeshCore does not load.  When using the console on the web flasher, commands to do not create any output or change in the device (set freq, ver, time, reboot).

I have attempted this on MacOS 15.1 and Windows 11.

recrof, 2025-02-28:
oh, just noticed the number.. we only support sx1262 t3s3.

fenixstarlord, 2025-03-03:
Is there a specific reason? I'm about to dive down the path of compiling my own firmware and wanted to make sure I'm not wasting my time.

recrof, 2025-03-03:
> Is there a specific reason? I'm about to dive down the path of compiling my own firmware and wanted to make sure I'm not wasting my time.

only reason is lack of time.. heltec v2 is supported and it has sx1276, so it might be pretty easy to do.

fenixstarlord, 2025-03-03:
Great! I'll give it a shot. I have no idea what I'm doing but no time like the present to learn.

closed 2025-03-03 by fenixstarlord