meshcore-dev/MeshCore #64: Hardware support for WaveShare RP2040 lora board.

imkermichel, 2025-03-05:
The board, not the hat module.

AeroXuk, 2025-03-16:
I'd like to also request the WaveShare Pico Hat as a variant, several Pico designs are based of this devices pinout.

AeroXuk, 2025-04-22:
PR #217 adds support for the Pi PicoW with WaveShare Lora Pico Hat.

@imkermichel could you send me a link to the module you have? If I have specs I might be able to add support for it.

imkermichel, 2025-04-22:
> PR [#217](https://github.com/ripplebiz/MeshCore/pull/217) adds support for the Pi PicoW with WaveShare Lora Pico Hat.
> 
> [@imkermichel](https://github.com/imkermichel) could you send me a link to the module you have? If I have specs I might be able to add support for it.

https://www.waveshare.com/rp2040-lora.htm

AeroXuk, 2025-04-22:
> https://www.waveshare.com/rp2040-lora.htm

Thanks, I've ordered one. It should be possible to add support with minimal change. Might take a couple of weeks for the module to arrive to me but I'll try to add support when it does.

446564, 2025-06-20:
Is this closed with #396?

liamcottle, 2025-08-06:
Added in #396

PR #396 (2025-06-18): Adds support for the Waveshare RP2040-LoRa board
Adds support for the [Waveshare RP2040-LoRa board](https://www.waveshare.com/rp2040-lora.htm).
Closes #64.
closed 2025-08-06 by liamcottle