meshcore-dev/MeshCore #66: power consumption

lafin, 2025-03-06:
Hi.
Yesterday I did some tests with meshcore and noticed that my powerbank was draining much faster than with meshtastic. I took some measurements (heltec v3 3.2):

meshcore: 167mAh
meshtastic: 94mAh

rightup, 2025-04-25:
Hi,
Just to clarify — what firmware variant were you running (e.g., Companion, Repeater)?
Also, during your tests, were both units connected via Bluetooth and left idle, to ensure an apples-to-apples comparison?

Thanks!

Delphinus44, 2025-05-17:
> Hi. Yesterday I did some tests with meshcore and noticed that my powerbank was draining much faster than with meshtastic. I took some measurements (heltec v3 3.2):
> 
> meshcore: 167mAh meshtastic: 94mAh

Did you use the same transmiting power? For example with Meshtastic 433 EU reduces the power according to the laws, Meshcore allow you to go to 22 in 433.

liamcottle, 2025-08-31:
Lots of changes have been made to firmware since this issue was opened. I believe there's been power optimisations for some devices. Will close out.

closed 2025-08-31 by liamcottle