meshcore-dev/MeshCore #76: Android app show NONAME on known and named device

michalsiman, 2025-03-07:
Heltec WSL3 - companion radio
Android app MeshCore v1.4.1

When again want to connect once connected and renamed device, by android app, shown in list MeshCore-NONAME, after connect shown correct name.

michalsiman, 2025-03-08:
Ok, Liam, than i am sorry. Than i cant restart my device, it battery powered and is in box and without any button or switch. But, when i would like a Reboot from app, i have a error message. Failed to reboot! i will wait for empty battery or unscruing my box for sure. But maybe error help for future improvement ...

[image]

recrof, 2025-03-08:
@michalsiman napis mi na recrof@gmail.com keby si potreboval podporu v cestine/slovencine.

closed 2025-03-12 by michalsiman