meshcore-dev/MeshCore #95: Feature Request:  Webflasher support for Pro-Micro DIY boards

thatSFguy, 2025-03-12:
I'm using these Washtastic boards : https://github.com/valzzu/meshtastic-pcbs/tree/main/WashTastic .  Currently I'm running Meshtastic on them with the NRF52 Pro-micro board selection from the web flasher.  What needs to be done to adopt this board to Meshcore?

I have no problem compiling the firmware like for the RAK wireless board, if needed. 

Thanks!
P.S. I just did some simple tests and I love how responsive the messaging is over MC!

recrof, 2025-03-12:
If you are adventurous, you can look at this diff and try to implement it yourself(this is basic support for t-echo).. https://github.com/ripplebiz/MeshCore/pull/68/commits/2dd2571154907f2fe4a4f3a5065ebac5f41ea95e

Hude06, 2025-03-13:
Hey @thatSFguy I have a few of the washtastic boards and im working on attempting to create a build for meshcore. Might take a few days. Ill let you know if i get anywhere with it

recrof, 2025-03-13:
> Hey [@thatSFguy](https://github.com/thatSFguy) I have a few of the washtastic boards and im working on attempting to create a build for meshcore. Might take a few days. Ill let you know if i get anywhere with it

drop on discord if you need any help!

Hude06, 2025-03-13:
Only thing im confused on is what I need to change. I think its only the pin out for varient.h and add one entry in the companion radio and in the platform.ini does that seem correct? Whats in the boards folder with the .jsons files for? Is that all the build flags? Do I need anything special for a nrf52 and a SX1262

recrof, 2025-03-13:
you don't have to worry about nrf52 and sx1262, just the json and variant files. sent you complete example for t-echo. should be very similar.

thatSFguy, 2025-03-13:
I got side tracked, so I have not even started yet.  I was going to just compare the data with the meshtastic info.  e.g. here is the JSON https://github.com/meshtastic/firmware/blob/master/boards/promicro-nrf52840.json  , or better yet https://github.com/search?q=repo%3Ameshtastic%2Ffirmware%20promicro&type=code . Not sure how close anyone else is on this.

recrof, 2025-03-13:
please don't use code from meshtastic, our licences are not compatible. for the json definitions, it might be fine, but be sure to not copy paste actual logic.

Hude06, 2025-03-14:
Just make sure you change the dbm in the meshcore app to 30 this code should theoretically work and he's working on submitting a PR

Hude06, 2025-03-14:
Yes, Just got it fully working on the Washtastic boards using the Fake Tec Varrient. This Issue can now be closed

thatSFguy, 2025-03-14:
I fell asleep on the couch waiting while working on this, but ran into an issue, so I'm glad @Hude06 got it done.  

One question, the washtastic board requires a setting of 22dBm in the "other firmware" based on output testing.  Is this calibrated in the MeshCore firmware?

recrof, 2025-03-14:
> I fell asleep on the couch waiting while working on this, but ran into an issue, so I'm glad [@Hude06](https://github.com/Hude06) got it done.
>
> One question, the washtastic board requires a setting [...]

not sure what "calibrated" means. radiolib has tx_power implemented - it's sent directly to semtech chip. build parameter `-D LORA_TX_POWER=22` should be enough.

thatSFguy, 2025-03-14:
> not sure what "calibrated" means. radiolib has tx_power implemented - it's sent directly to semtech chip. build parameter `-D LORA_TX_POWER=22` should be enough.

On the Station G2 a setting of 30(dBm) gets you 30dBm TX (1W).  However, with the Ebyte E22 chip, a setting of 22(dBm) gets you 29.37dBm output (almost 1 Watt), which is the top limit in the US.  Since you said it's sent directly to the chip (no conversion of any sort), I will continue to use the 22 setting for this board.  Thanks!

[image]

thatSFguy, 2025-03-16:
Final UPDATE:
- Both Washstatic and Faketec fall under the ProMicro variant, as credited by Faketec: 
 "The variant & design is based on the DIY [ProMicro variant]",  so for Meshcore you can use the Faketec build.
- The companion BLE is working
- The repeater seems to be not working.  I'll open a separate issue for this.

closed 2025-03-16 by thatSFguy