meshcore-dev/MeshCore #103: FakeTec v4

sebikolo, 2025-03-15:
Does Meshcore work on NRF52840 + FakeTec v4?

recrof, 2025-03-15:
https://github.com/ripplebiz/MeshCore/pull/90

thatSFguy, 2025-03-16:
Confirmed https://github.com/ripplebiz/MeshCore/pull/90 working with the Washtastic board as well (repeater and companion BLE).  

I would like to suggest to rename the FakeTec firmware to ProMicro, since leaving it like it is, will cause confusion.  Even FakeTec calls their board a ProMicro variant.  Thanks!

One last thing, the build.sh script still needs to be updated to build the uf2 file for promicro  in the dev branch.   
`  # build .uf2 for nrf52 boards
  if [[ $1 == *"RAK_4631"* || $1 == *"t1000e"* || $1 == *"t114"* || $1 == *"T-Echo"* || $1 == *"promicro_nrf52840"* ]]; then
    python bin/uf2conv/uf2conv.py .pio/build/$1/firmware.hex -c -o .pio/build/$1/firmware.uf2 -f 0xADA52840
  fi`

sebikolo, 2025-03-18:
Can you share uf2 file for NRF52840 + FakeTec v4?

thatSFguy, 2025-03-18:
Here is the one for the repeater.  Not sure where I put the companion one, so I would have rebuild it.  https://github.com/thatSFguy/MeshcoreBuilds

thatSFguy, 2025-03-19:
> Can you share uf2 file for NRF52840 + FakeTec v4?

Added the companion ble one as well now

sebikolo, 2025-03-22:
Yes :-)

closed 2025-03-23 by ripplebiz