meshcore-dev/MeshCore #106: [Feature Request] Get meshcore:// map link from serial console

clintonblackmore, 2025-03-15:
It would be convenient, especially for users who don't have any nodes that connect to the app, to be able to get the meshcore:// URL to be able to put onto the serial console.

Ideally, in the console, the user would type something like "get link" and get an answer back like "meshcore://a5f3...." with the full value they need to use to put their node on the map.

recrof, 2025-03-15:
> It would be convenient, especially for users who don't have any nodes that connect to the app, to be able to get the meshcore:// URL to be able to put onto the serial console.
>
> Ideally, in the [...]

you can avoid this by sending an repeater advert to the companion device and copying meshcore link from there.

[image]

milgrim2023, 2025-04-08:
the problem comes if your companion device is a t-deck that doesn't connect to the app. Unless I'm missing something?

recrof, 2025-04-08:
for t-deck, there is another solution: https://github.com/ripplebiz/MeshCore/blob/main/docs/faq.md#411-q-what-is-the-import-from-clipboard-feature-on-the-t-deck-and-is-there-a-way-to-manually-add-nodes-without-having-to-receive-adverts

recrof, 2025-04-08:
you can save any contact to sd card, even your repeater.

milgrim2023, 2025-04-08:
From the t-deck? 

I don't see any UI flow that lets me do this. Selecting from 'Discover' lets me show on map, reset advert clock, remove from home, and share..none of which lead to a save to clipboard option.

Apologies if I'm missing something obvious here.