meshcore-dev/MeshCore #108: Info-Room Server - read only mode

kurtrenauer, 2025-03-16:
I have a "feature-request" that would give me an interesting additional use case. I suggest to implement a read only mode for room servers to "visitors". It could be implemented e.g. without password you can enter the read-only mode.  My use case is the idea of an info-room for visitors to keep it clean. 
Liam Cottle added the idea of having a flag or something in the room advert would make it easy for clients to show a banner saying it's read only, and hide the send message inputs.
What do you think?

ripplebiz, 2025-03-16:
Yes, that would definitely be useful.
I'll add it to the to-do.

pdomenig, 2025-03-16:
Isn't this a duplicate of that one? https://github.com/ripplebiz/MeshCore/issues/75

chixxi, 2025-03-16:
Isn't this a duplicate if that that one? https://github.com/ripplebiz/MeshCore/issues/75

kurtrenauer, 2025-03-16:
@chixxi ... oh yes, that is way the same request.  it is just the wording that's different, but the same result I guess.

closed 2025-03-17 by ripplebiz