meshcore-dev/MeshCore #113: Station G2 issues

Martje63, 2025-03-16:
Station G2 issues with 1.3 meshcore-firmware
No display in repeater mode
No display in Roomserver mode
No possibility to log in, standard password does not apply.
Bluetooth in both webclients do not work - laptop W10
Serial access does not work, (possible cause: Laptop does not support 15V on usb-c port, 15 V needed for Station G2

Martin pd0zz

recrof, 2025-03-16:
display is not implemented in any role, will get implemented later.
Serial access should work even when you're connected to the 5V usb. Only thing that doesn't work when you use 5V is PA - everything will get TXed, but without extra power. did you restart Station G2 after flashing?

recrof, 2025-03-16:
> Bluetooth in both webclients do not work - laptop W10
you can't connect to repeater / room server via bluetooth. you can administer them from serial console or using companion device and app via LoRa.

Martje63, 2025-03-16:
OK, but the standard password for admin connect to **Station G2 Room** via android app does NOT work , no 123456 login... How to solve that problem?

recrof, 2025-03-16:
standard default password for managing the room server is `password` and access for users is `hello`.

Martje63, 2025-03-16:
Can be closed, could not find the correct password, solved! 
Other issues solved for now! 

Thanks for the answers!

closed 2025-03-16 by Martje63