meshcore-dev/MeshCore #115: [Feature request] Disabling BLE Pin

awolden, 2025-03-17:
For some automation needs like connecting to [HA](https://github.com/awolden/meshcore-ha) through [bluetooth proxy](https://esphome.io/components/bluetooth_proxy.html), the node can only be connected over bluetooth if the pin is disabled. It would be nice to be able to optionally disable the the pin in the companion radio firmware.

recrof, 2025-08-06:
here is a patch that you can use until we find permanent solution: https://gist.github.com/mannkind/9290f64f815750147d78857dbc110606

liamcottle, 2025-09-16:
Closing this one out since a lot has changed in the firmware over the last 6 months. Don't think we added support for no BLE pin, but I've heard the home assistant setup is doing well, so I'm assuming this isn't an issue anymore.

closed 2025-09-16 by liamcottle