meshcore-dev/MeshCore #116: Some Adverts Displaying As <Unknown Name>

mikewren, 2025-03-17:
I received two adverts today indicating <Unknown Name> on a RAK node connected via BLE to v1.5.0 of Liam's app on Android.  Source node is a Station G2.

All nodes running v1.3.0 firmware.

Here's the MeshCore link for the G2 as sent: meshcore://1200e24735cfec3549b1efde6a9130aae01ba5d612f42a620ac3d20e859eae6b51e3fd81d8677c39a306019cfc37bf57071263561632f4d5f7581f129b1d7306a86bb750140d7ee7cf566c4ab2efcdff1e605905282525cd13c05b6cd8a2327963575deb9906932fd98a0221849afb4361706974616c204469737472981b0120a81a0120

{
  publicKey: "e24735cfec3549b1efde6a9130aae01ba5d612f42a620ac3d20e859eae6b51e3",
  timestamp: 1742242301,
  isVerified: false,
  advLat: 42.653999,
  advLon: -73.759711,
  advName: "Capital Distr�\x1b\x01 �\x1a\x01 ",
  type: 3
}

Node name is "Capital District Mesh"

recrof, 2025-03-19:
this should be resolved - those were caused by bug in the ed25519 library we used <v1.4.0

liamcottle, 2025-03-19:
We discovered a fun memory corruption bug in firmwares v1.3.0 and older, when verifying the advert signature.

The ed25519 verify function has been swapped out for another library and it seems to have resolved the bug.

Lots of debugging was done on this issue, and it was crazy that it was an external lib causing it...

Please update your device firmware to v1.4.0 and let us know if you notice it happening again :)

closed 2025-03-19 by ripplebiz