meshcore-dev/MeshCore #119: Range Testing Mode [feature Request]

Soupbones, 2025-03-19:
A fundamental part of setting up mesh networks is to test how your signal gets out to the locations you want it to. Meshtastic has range testing mode but the way you use it, like many things on Meshtastic is overly complex and bloated to how it should be. It would be great to see Meshcore implement a simple and easy-to-use range test mode and implement it into the mobile app only. Range test would be set up so that its all run through the mobile apps in something like the following way.
1. Open up Range test mode, and select the repeater that you have remote access to. Enter the password and you enter the remote repeaters Range test settings. Here you would put the repeater into Range test mode, as well as the companion node you are testing. 
2.  First thing the app will ask is for the user to set a different frequency, This frequency being set will restrict you to a frequency either above or below the previous operation frequency that is already set in both the repeater and companion node by at least 500kHz difference, so as not to cause interference to the main usage frequency. 
3.  Then you set an interval for the range test signal from the repeater. This can be a setting from 1 min up to once every ten mins. 
4. Once these simple parameters are set up, the repeater stays on the new adjusted frequency as the companion node which is now also in standby mode for range testing.
5. Once you are ready to go mobile, or reach your first range test destination, you can start the range test communications between the repeater and companion node using a start button. The companion nodes' location will be saved to a database along with the signal strength of the repeater. As you go mobile it will log any signal hits it gets onto the map and store the information in the database. This signal strength would be simply displayed as a number between 1 - 10 at each hit on the map. Pressing any of these range test numbers will give you the RSSI and SNR as recorded in that hit.
6.  Once range testing is complete, you hit stop and stop on the app to halt the range testing. This must be done with the repeater in range. This will send the repeater back to the main frequency as well as put the companion node back to the main operation frequency as well. The app will then confirm the repeater and node are back to a standard setting.
7. The stored data can be then selected and viewed on the map with a simple icon for each time hit with a number. There will also be the option to export this in CSV or Google Maps format to the clipboard.
 I know Meshcore has a long journey ahead, but its features like this that will assist in setting up networks

liamcottle, 2025-03-19:
The newly released v1.4.0 firmware has a new TRACE packet, which allows for collecting SNR for a given path. The mobile app now has a custom path trace tool, as well as a new map based path tracing tool. Please update all of your repeaters and companion nodes and give it a go. :) Keen to hear any feedback you may have.

At this stage, there is no automated testing for a set interval, for now you click the button and it will do a single trace per click.

I may implement a coverage testing tool a bit later down the track, as I too would like to test where my repeaters are accessible from, and what their average signal strength is at any given gps position.

Soupbones, 2025-03-19:
Thanks Liam. A coverage testing tool with map export would be great. I just put up a repeater today running 1.20 so it will have to wait a bit. I found the RAK in repeater mode has no Blue Tooth so its a pain to go and update it.

Emmpunkt, 2025-03-19:
Use remote management to log into your repeater. (As Admin)
Tap the "Command Line" Button.
Type in "start ota", enter.
Use nRF Connect to update your repeater....

michalsiman, 2025-03-19:
what is "nRF Connect" for OTA update? when i googling, there is lots of apps with this name, can you please let us know and can you maybe wrote little bit more info about how to flash via OTA? Thank you so much.

recrof, 2025-03-19:
> what is "nRF Connect" for OTA update? when i googling, there is lots of apps with this name, can you please let us know and can you maybe wrote little bit more info about how to flash via OTA? Thank you so much.

please read the FAQ: https://github.com/ripplebiz/MeshCore/blob/main/docs/faq.md#q-how-to--update-repeater-and-room-server-firmware-over-the-air

Soupbones, 2025-03-19:
Will OTA flashing this way over write the operation freq? I'm told the OTA code has the EU/UK freq hard coded in.

recrof, 2025-03-19:
> Will OTA flashing this way over write the operation freq? I'm told the OTA code has the EU/UK freq hard coded in.

default hard-coded frequency is always overriden by your saved radio parameters. only time where it might present a problem is when changing roles. it doesn't matter if you use flasher or nRF OTA updates.

Soupbones, 2025-03-19:
Thanks for that. Another Mesh user I know was concerned about the hard-coded freq and was changing it for 915.800Mhz here in Australia in the code. I'm no coder so I don't want to do the hard yards to access a remote node only to lose access to it after not changing the freq.

closed 2026-01-18 by liamcottle