meshcore-dev/MeshCore #120: BUG: node_prefs file is not compatible between roles

recrof, 2025-03-19:
Currently, all roles share same file `node_prefs`, but the format is not the same, which causes problem when re-flashing different role - radio settings get mangled. we need to either keep "shared" config with same format, or save different name for each role.

sevenbitbyte, 2025-03-19:
It'd be really nice to use an extensible format. Either some text formet (eg: .ini, .json) or a binary format (eg: protobuf, bson).

recrof, 2025-03-19:
> It'd be really nice to use an extensible format. Either some text formet (eg: .ini, .json) or a binary format (eg: protobuf, bson).

we are talking about internal flash file that is not user-accessible.

sevenbitbyte, 2025-03-21:
I'm not suggesting that as a user accessiblity feature, I'm suggesting it because the `node_prefs` can become incompatible between the same role as we add more features and make changes.

recrof, 2025-03-21:
using text file format would be pretty wasteful for both clock cycles and storage space(look how much space you have on nRF52 boards). binary format is way to go.

sevenbitbyte, 2025-03-22:
What boards are you specifically worried can't handle an extensible file format? I see it's typical for nrf52's to have around 1MB of storage space. I don't see why json, bson or protobuf wouldn't work equally well on nordic platforms.

I see a lot of well tested arudion focused libraries with great performance and even some are actively tested on nrf52 & PlatformIO like ArduinoJson:

https://github.com/bblanchon/ArduinoJson

There's also bson libraries like ArduBSON which are already published on PlatformIO and appear fairly portable:

https://github.com/argandas/ardubson

recrof, 2025-03-22:
I'm not worried it would not handle text formats, I'm saying it's wasteful. in case we want to use the space for something else, we would regret going text route. 1MB is pretty scarce. I'm not arguing about other more portable binary formats.

closed 2025-03-30 by recrof