meshcore-dev/MeshCore #125: [Feature Request] Add 433 MHz preset and disable encryption

BadCactus634, 2025-03-20:
Hi, I am the founder of LoRa Brescia, a group that helps people from the Brescia province (the biggest in Italy) getting started with LoRa projects.

I am also part of the Italian Civil Protection and I want to ask my supervisor about implementing an emergency network based on MeshCore on 433 MHz band as we are all ham radio operators and so have the license to use that band, with all the benefit that it provides, giving that it's less susceptible to noise that the 868 band.

In the app I can't find any preset made for that band, so I have to tell everyone to get to the same frequency.

I’m askink you if it is possible to add a preset in the app for the 433 band for us radio ham operators, so we are all on the same page about the radio configuration.

Also, I can’t find any infos about the encryption: is it possible to disable that entirely? Ham radio operators must transmit unencrypted traffic.

fredzo, 2025-03-25:
Hi @BadCactus634, same here in France, we could use MeshCore for ham radio emergency communications (Civil Security operators)  on ham bands, but we need unencrypted messages for regulatory reasons.

RE-Specto, 2025-04-08:
definitely a "licensed amateur radio operator" checkbox similar to meshtastic would be useful.
however btw for the regulation, afaik most of the countries rule doesn't mention explicitly the word encryption, but rather "prohibits any type of operation if the purpose is to obscure the meaning of your transmission" and while many interpret this as a ban on encryption, actually if you for example publish your keys (or use a "default" key for your channel) legally there is no "purpose to obscure" anymore and it can be considered an encoding rather than encryption.

BadCactus634, 2025-04-08:
Right, I think this is remarkable: disabling encryption provides even simpler configuration. Also, that means less channel utilization as there is no key that needs to be transmitted with the data every time.

I again have to point out that **having a default preset for the 433 band is mandatory**, not everyone uses MeshCore on 868 MHz.
There are many people interested in using 433 band, but the lack of this preset makes setup unnecessary hard because you have to contact each node owner to tell him to use that frequency.