meshcore-dev/MeshCore #130: SenseCAP T1000 freezes when it receives a 160-character message

sdqsds23213, 2025-03-21:
When I send a text message with the maximum number of 160 characters to a SenseCAP T1000, the tracker receiving the message disconnects from Bluetooth and freezes (steady yellow LED). I then have to force a reboot by connecting it to its power cable. On the sender's side, the message is marked as not received on the sender's Android application.

Sender and receiver are both SenseCAP T100s, with firmware 1.4.0.

Problem repeated when testing in the other direction.

Thanks for your hard work.

sdqsds23213, 2025-03-21:
Thanks, I will try to install it !

sdqsds23213, 2025-03-21:
@kt-repp I tried with this custom firmware, there is the same problem...

fdlamotte, 2025-03-21:
Thanks for the report, I'll investigate !