meshcore-dev/MeshCore #132: T-echo screen

pacevedos, 2025-03-21:
Hi! i just flashed my T-Echo with the 1.4.0 and the display is not working. I also tried with the 1.3.0 and the results were the same.

From what I've read I think it might not have been implemented yet but I just wanted to let you know in case this is not something you were aware of.

Thank you for your work!

recrof, 2025-03-21:
yep, the screen is not implemented. we are focusing on more important features first.

frantsch, 2025-04-11:
would be great if it would be at least in a simple way, though!

recrof, 2025-04-11:
> would be great if it were, at least in a simple way, though!

it'll won't be long now @fdlamotte *wink* *wink*

frantsch, 2025-04-12:
This would  be very usefull for all us t-echo users, I encourage you

pacevedos, 2025-04-21:
It works! Thank you!

fdlamotte, 2025-04-21:
Thanks for the report, always good to know when it works and it is used ;)

I have some cool ideas for this screen as it can show a lot of information ... 

Already found a misbehaviour (won't call it bug), I let people find what it is, it is not harmfull ;)

closed 2025-04-22 by pacevedos