meshcore-dev/MeshCore #133: Battery Level adjustment

Emmpunkt, 2025-03-22:
Is it possible to adjust the battery level?
My Heltec V3 shows 71% when the battery is fully charged (4.176V)
Thanks M.

liamcottle, 2025-03-23:
Battery percentage is calculated in the web client and mobile client, the firmware only provides the voltage.

At this stage, the calculation is linear, and I didn't put too much time into tuning it. We can probably add some sort of table based on different boards, but for now, this is the currently calculation code:

```dart
class BatteryUtil {

  static int getBatteryPercentage(int millivolts) {

    const minVoltage = 3400; // show battery as 0% at or below this value
    const maxVoltage = 4200; // show battery as 100% at or above this value

    // 0% if at or below min voltage
    if(millivolts <= minVoltage){
      return 0;
    }

    // 100% if at or above max voltage
    if(millivolts >= maxVoltage){
      return 100;
    }

    // linear calculation
    // todo implement curve based voltage to percentage calculations
    return (((millivolts - minVoltage) / (maxVoltage - minVoltage)) * 100).floor();

  }

}
```

Emmpunkt, 2025-03-23:
Thanks for the answer,
but i don´t think that this is the reason.
I think the value from the Heltec is wrong and needs adjusting.
Even with USB connected and full battery it does not show more then 71%.
Can i check the voltage comming from the Heltec somewhere? CLI?

recrof, 2025-03-23:
@Emmpunkt you can check it here: https://liamcottle.github.io/meshcore.js/ (connect and choose GetBatteryVoltage button)

Emmpunkt, 2025-03-23:
OK,
Battery voltage measured : 3744mV (Kaiweets HT206D)
Voltage by the device : 3552mV (16%)

deadphyre, 2025-03-23:
I just wanted to comment on here as I'm using meshcore on the same device and also see that it does not go above 71%. Not that this helps any but does confirm Emmpunkt's discovery.

recrof, 2025-03-23:
[Heltec_v3_companion_radio_ble-dev-merged.bin.zip](https://github.com/user-attachments/files/19410837/Heltec_v3_companion_radio_ble-dev-merged.bin.zip) please try this firmware - it has alternative voltage retrieval routine.

Emmpunkt, 2025-03-23:
@recrof Battery is now at 0mV and 0% ;)

Emmpunkt, 2025-03-26:
Can someone please change Line 96 in HeltecV3Board.h
from:
`return (5.2 * (3.3 / 1024.0) * raw) * 1000;`
to:
`return (5.42 * (3.3 / 1024.0) * raw) * 1000;`

I found the number in the Heltec Forum...
Thanks M.

recrof, 2025-03-31:
> Can someone please change Line 96 in HeltecV3Board.h from: `return (5.2 * (3.3 / 1024.0) * raw) * 1000;` to: `return (5.42 * (3.3 / 1024.0) * raw) * 1000;`
> 
> I found the number in the Heltec Forum... Thanks M.

you can open PR if you believe those are accurate for all V3s.

Emmpunkt, 2025-03-31:
I´m not a programmer, i´m (industrial) electrician.
It would be better if the value is adjustable because it depends on 2 resistors with unknown quality.
In my case the 5.42 is way better then the 5.2, but there is still room for more accuracy...

Emmpunkt, 2025-06-20:
At least it killed my config.... And i don´t have a Backup...
I have to charge my battery first to tell more.

446564, 2025-06-20:
> At least it killed my config.... And i don´t have a Backup... I have to charge my battery first to tell more.

Apologies for that 😢

446564, 2025-06-20:
nice, ya I found it was accurate on my wsl

PR #423 (2025-06-21): heltec v3 update ADC multipler to fix voltage reading
seems to work fine on my WSL3

closes #133
closed 2025-08-31 by liamcottle