meshcore-dev/MeshCore #136: T-Beam No Bluetooth and no Serial

grahamrobertslx, 2025-03-23:
Hi there, 

Moved My T-Beam over from another Mesh firmware, But when flashing it completes, but isnt viewable on bluetooth and cant get into it via serial. I am able to re-flash with the other Mesh firmware and works absolutely fine. T-Beam is a v1.1 with neo-m8n power chip. Sorry i cant give any more info. Happy to try some things

jameshall9327, 2025-03-24:
Unpair it from your phone if it was previously paired. Also, try to erase the device with the flashed and see if that works. I had that problem and that fixed it

> Hi there,
>
> Moved My T-Beam over from another Mesh firmware, But when flashing it completes, but isnt viewable on bluetooth and cant get into it via serial. I am able to re-flash with the other Mesh [...]

grahamrobertslx, 2025-03-24:
Arrr standby caller, found the second T-Beam bin file. One is merged and one isnt. Not sure what the Merged is. Manually uploaded the non Merged version and bluetooth has sprung to life. Looks like the Web Installer only has one version

closed 2025-03-25 by grahamrobertslx

tbaumann, 2025-09-18:
Switching on all DEBUG for  `Tbeam_SX1276_companion_radio_ble` I can see the device is working. The real issue is the blank Display.

Looking for `BLE: onPassKeyNotify(213038)` on the serial monitor allowed me to see the BLE pin code (213038) and bind.

I will investigate further what might be the cause for the display differences. I thinkperhaps a power rail on AXP2101 is not brought up...