meshcore-dev/MeshCore #140: t-beam doesn´t received messages

Atorcha, 2025-03-25:
Hello, I have a TTGO Lora v2 linked with an IOS (iPhone) and a T-BEAM linked with an android.

First I flashed TTGO, then I flash T_BEAM.

I send "Advert (ZERO Hop)" with T_BEAM and received the node in TTGO. 
I send "Advert (zerohop)" with TTGO and NOT RECEIVED the node in T-BEAM.

They are 2 meters separeted.

jameshall9327, 2025-03-26:
Could be the code of the ttgo

Atorcha, 2025-03-27:
I think so, because with meshtastic it works

davehun, 2025-04-04:
I can confirm that the latest release does not work but the latest git compiled and loaded onto a device works

jameshall9327, 2025-04-04:
> I can confirm that the latest release does not work but the latest git (259f4ec1697920cf8d279455a24d20663d8628c9) compiled and loaded onto a device works

There are lots of variants of the T-Beam. Which one do you have?

davehun, 2025-04-04:
> > I can confirm that the latest release does not work but the latest git ([259f4ec](https://github.com/ripplebiz/MeshCore/commit/259f4ec1697920cf8d279455a24d20663d8628c9)) compiled and loaded onto a device works
> 
> There are lots of variants of the T-Beam. Which one do you have?

t22_v1.0

davehun, 2025-04-04:
> > > I can confirm that the latest release does not work but the latest git ([259f4ec](https://github.com/ripplebiz/MeshCore/commit/259f4ec1697920cf8d279455a24d20663d8628c9)) compiled and loaded onto a device works
> > 
> > 
> > There are lots of variants of the T-Beam. Which one do you have?
> 
> t22_v1.0

the 1.4.2 didn't send or receive any messages successfully on that device

jameshall9327, 2025-04-04:
> 
> 
> > > I can confirm that the latest release does not work but the latest git ([259f4ec](https://github.com/ripplebiz/MeshCore/commit/259f4ec1697920cf8d279455a24d20663d8628c9)) compiled and loaded onto a device works
> > 
> > There are lots of variants of the T-Beam. Which one do you have?
> 
> t22_v1.0

The one that is supported is the tbeam 1.2 with the SX1276

davehun, 2025-04-04:
> The one that is supported is the tbeam 1.2 with the SX1276

I *think* it has the same chip
output from terminal during a boot (when TX and RX works i.e. current main

```
 pio  run   -t monitor --monitor-port  /dev/tty.SLAB_USBtoUART 
Processing Tbeam_companion_radio_ble (board: ttgo-t-beam; platform: espressif32; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Please build project in debug configuration to get more details about an exception.
See https://docs.platformio.org/page/projectconf/build_configurations.html

--- Terminal on /dev/tty.SLAB_USBtoUART | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
RLB_DBG: 
RadioLib Info
Version:  "7.1.2.0"
Platform: "ESP32"
Compiled: "Apr  4 2025" "19:56:00"
RLB_DBG: M	SX127x
DEBUG: App meshcore-flutter connected
: TX, len=108 (type=4, route=F, payload_len=106)
: RX, len=22 (type=2, route=D, payload_len=20) SNR=10 RSSI=-40 score=1000 hash=8D3758D1B3F924A4 [CA -> 7C]
: TX, len=6 (type=3, route=D, payload_len=4)
: TX, len=22 (type=2, route=D, payload_len=20) [7C -> CA]
: RX, len=6 (type=3, route=D, payload_len=4) SNR=9 RSSI=-29 score=1000 hash=DEE58B802448B145
```

closed 2025-08-31 by liamcottle