meshcore-dev/MeshCore #146: Companion BLE crashes after sending long channel message

oltaco, 2025-03-27:
The companion BLE firmware will crash at some point after sending a long channel message of at least 153 chars.

Device: Heltec v3.2 / Faketec (Both Heltec, both Faketec, any combination thereof). Probably others?
Firmware version: Companion BLE 1.4.1 main and dev branch.

Once the crash is "set up" it will take another action to cause the crash to trigger. One way is to login to a repeater and choose get firmware version, the app will show this times out and the companion BLE will crash (and reboot in the case of Heltec v3.2, but Faketec will just hang).
Another way is to choose disconnect in the app, and then reconnect to the companion. It will connect but shortly afterwards the crash is triggered same as before.

Steps to reproduce:
1. Send a channel message of at least 153 chars (or less depending on node name). Now the crash is "set up".
2. Either login to repeater and choose get repeater firmware version, or disconnect from the companion BLE and reconnect. 
3. The device will crash, Heltec will reboot but Faketec does not.

It's at least partly related to how long the BLE node name is, because the the channel message length required to set up the crash becomes shorter with how long your BLE node name is. For example if your node name is "1" the crash will be set when you send a channel message of at least length 153 chars, but if your node name is "123456" the crash will be set when you send a channel message of at least 148 chars.

It's also directly related to MAX_FRAME_SIZE as set in BaseSerialInterface.h, because if you bump this up to 185 the error in the BLE log goes away, and the crash no longer happens. However having said that, there's still something funky going on with node name length because even with MAX_FRAME_SIZE set to 185 you still can't send channel messages of above a certain size, and those channel messages are still truncated on receiving clients. And above a certain size the app will even complain that it couldn't send the message.

I've attached logs from companion BLE and repeater for each of the crash triggers I found.

[01 CompanionBLE crash triggered by request repeater version.txt](https://github.com/user-attachments/files/19480078/01.CompanionBLE.crash.triggered.by.request.repeater.version.txt)
[01 Repeater log.txt](https://github.com/user-attachments/files/19480079/01.Repeater.log.txt)
[02 CompanionBLE crash triggered by reconnect BLE.txt](https://github.com/user-attachments/files/19480076/02.CompanionBLE.crash.triggered.by.reconnect.BLE.txt)
[02 Repeater log.txt](https://github.com/user-attachments/files/19480077/02.Repeater.log.txt)

oltaco, 2025-03-27:
Attached is a coredump I managed to get a few days back. It's from heltec v3.2 running **firmware 1.4.0** after a crash triggered by request repeater version. Not sure why my Heltec is no longer giving coredumps with this crash unfortunately.

[heltec v3.2 coredump req repeater ver.txt](https://github.com/user-attachments/files/19493514/heltec.v3.2.coredump.req.repeater.ver.txt)

recrof, 2025-03-27:
> Attached is a coredump I managed to get a few days back. It's from heltec v3.2 running **firmware 1.4.0** after a crash triggered by request repeater version. Not sure why my Heltec is no longer [...]

Are you able to get the crash in controlled enviroment? if you could in vscode platform.io do upload + monitor to get a proper stacktrace, that would be big help. maybe keep it on for a longer period of time? thank you for your help!

oltaco, 2025-03-28:
Did some testing with latest dev commit https://github.com/ripplebiz/MeshCore/commit/c0cb57be5c5c08774549cd7ee304eae8755b5af7 on Heltec V3.2 with BLE_DEBUG only.

Crash behaviour is different depending on length of nodename. 

If nodename is 1 or 2 characters you'll get a panic with stacktrace. (attached) 
If nodename is 3 4 or 6 chars, corruption happens but not a full panic, timeouts occur app when requesting repeater info, device stays active on BLE, radio is still working (lights flash and repeater receives and replies), firmware seems to not realise that the repeater responses aren't normal messages, so it displays them on the screen.
If nodename is 7 chars corruption happens and you get a couple of timeouts in the app but it seems to resolve itself and then continue working.
If nodename is 11 chars you can trigger a panic with reconnect to BLE but no stacktrace.

[coredump 1char nodename.txt](https://github.com/user-attachments/files/19496510/coredump.1char.nodename.txt)
[coredump 2char nodename.txt](https://github.com/user-attachments/files/19496509/coredump.2char.nodename.txt)

liamcottle, 2025-03-28:
For extra context:

- Max node name length is [32 characters](https://github.com/ripplebiz/MeshCore/blob/88b88cbc901f2a1dd5329f84901dde4546d82c44/examples/companion_radio/main.cpp#L184). 
- Max group message length is [160 characters](https://github.com/ripplebiz/MeshCore/blob/88b88cbc901f2a1dd5329f84901dde4546d82c44/src/helpers/BaseChatMesh.h#L8).

ripplebiz, 2025-03-28:
OK, I think the problem was the logRxRaw().
I forgot that the _raw_ packet length can be longer than MAX_FRAME_SIZE :-(
Have pushed a temporary workaround to dev (for now). Let me know if the crash goes away now (It *should*)
I'll have to analyze the impact of increasing the MAX_FRAME_SIZE, or some redesign of the PUSH_CODE_LOG_RX_DATA frames.

oltaco, 2025-03-28:
Yes the workaround does the panic from my quick testing. 👍

I'm guessing at the cost of missing the too long frames from the log though?

I didn't find any problems when I increased MAX_FRAME_SIZE to 185 during my testing but I didn't want to submit a PR since I'm not aware of what the implications might be, best left you.

I think there's still a separate issue with regard to channel messages being truncated on the receiving end, but I'll do some more testing and submit a separate issue if I turn something up.

ripplebiz, 2025-03-28:
So, it still crashes?

ripplebiz, 2025-03-28:
Ah, sorry I misread your comment. So, workaround does work.

oltaco, 2025-03-28:
Sorry, I missed the most important word before I edited it! 😢

closed 2025-08-31 by liamcottle