meshcore-dev/MeshCore #147: T114 Screen use

lucvan, 2025-03-27:
At present the screen on a T114 is blank and misses the same status features of the V3 etc. Can these features be implemented in the T114 firmware?

inversenebula, 2025-04-06:
Evidently according to a discussion a few days ago on Andy Kirby's Discord, one or more people have been working on this, but it might be down to one person now (or none). March 22nd advice was that the display is "a bit tricky" and "please be patient." So, it's unclear. Hopefully someone who knows more will comment here.

recrof, 2025-04-21:
tt14 is currently in experimental stage and can be user-built. there is one worry, that the speed of repaint using stock arduino lib might be too slow and might interfere with radio timings. here is sample snapshot with 1.5.1 screen build: https://flasher.meshcore.dev/firmware/Heltec_t114_companion_radio_ble_screen.zip (use web flasher > custom firmware to flash)

446564, 2025-06-20:
closed in #402

closed 2025-08-06 by recrof