meshcore-dev/MeshCore #150: Google Pixel Devices Cannot Resolve Bluetooth PIN Code

ManCityFC, 2025-03-28:
No idea why this happens, but it is a known bug with Pixel Phones. When you try to connect via Bluetooth, it won't connect. The options to select the PIN flash on the screen and back off again so quickly you can't accept the PIN. Could you add an option to connect without a PIN please?

jameshall9327, 2025-03-28:
> No idea why this happens, but it is a known bug with Pixel Phones. When you try to connect via Bluetooth, it won't connect. The options to select the PIN flash on the screen and back off again so quickly you can't accept the PIN. Could you add an option to connect without a PIN please? 

I had that same issue with my samsung s24 ultra. It ended up being another app on my phone that uses bluetooth that wasn't even running

ZilvinasJonaitis, 2025-04-22:
I have Pixel 7 Pro and Pixel 8a. Both run Android 15. I have not experienced such an issue. However, this the same or very similar problem, when trying to connect via Web Client on PC. As soon an PIN entry windows is presented, there is no way to recall PIN number on Heltec V3 screen via button push. PIN is not displayed there anymore.

closed 2025-09-16 by liamcottle