meshcore-dev/MeshCore #156: Meshcore - Map Radio params

Emmpunkt, 2025-03-30:
Because of the recently change in the default frequency from 867.5 to 869.525 Mhz,
how can i set the "Radio params" in the Meshcore-Map?
In the moment i can only reach my own devices. All other stations are too far away (Germany, hilly terrain).
How can they know that the frequency has changed? Especially if they did not update their devices?

juergen908, 2025-03-30:
Hello Emmpunkt.
I am near to you (Bruechermuehle) and drove fiew days ago to Nümbrecht and tried to connect you...
and the Problem have been the "Radio-Params" i didnt know !
So its a good idea to see them in MeshCore-Map.
By this way anyone can connect to the Oberberg-Net we build up...

Emmpunkt, 2025-03-30:
@juergen908 Next week my mobile repeater is in Brüchermühle between 13:30 and 21:00. If i can connect your "Room", i will leave my E-Mail or we can talk directly... I also tried to connect you, but with the wrong frequeny. But i had a device named "Client Keller" in my list?!

juergen908, 2025-03-30:
Emmpunkt:
Wonderfull.
We are running at 
Freq: 869.525,
BW: 250,
SF: 9,
CR: 6

Please delete my nodes in your system.
They are corrected today in Internet-Map.

My Roomserver
meshcore://1100066559f03c17275a4cf1330131f6a55aa08146e81ef7365870e14553820e96f94cf6e8674ebe08bf0048b30992dc60e9df4fa551ea5843a74c3801075371a392a44aac56c3b21dc2b23cafe8a2d61bcdd7b34e0ab488b372714ed0613643af5913a0e70a93560a090337907400526f6f6d2d42727565636865726d7565686c65

My Client
meshcore://1000992a88feb4251fc5b8b1a1d52ce81203b53727726078f17918a90260b68edfe8fef4e8674303af766850794e04bd4f393e90d6ccf688621bd586452142decbe8824fb3565920f90d325d27e7bf478f67fc14e02663a67bd87803130d979e4069506ebc0191560a090343917400436c69656e742d706f72746162656c

recrof, 2025-04-22:
@Emmpunkt this was implemented on the map and app side.. if you upload nodes from the app, radio settings are populated. if you're satisfied with it, please close the issue.

recrof, 2025-09-01:
> The radio params isn't populated on the map for my repeater. It's a LILYGO T-Beam v1.2 running v1.7.4-c1041af (Build: 24-Jul-2025).

how are you uploading the node to the map?

hofmand, 2025-09-01:
I did it from the Android App, "Copy Link to Clipboard" and then pasted it into the map. But just now after I saw your question, I tried "Upload to Internet Map" and that populated the radio params field. I didn't realize there was a difference between the two methods. Thanks!