meshcore-dev/MeshCore #159: [Feature Request] Ability to shut down nodes

michaeljaddy, 2025-03-31:
Please would it be possible to implement the ability to shut down nodes to allow for safely changing antennas etc without having to disassemble the node to disconnect the battery.

Many thanks

Mike

juergen908, 2025-03-31:
Maybe a switch in App to disable TX...

neutralinsomniac, 2025-04-12:
So is it normal for there not to be a shutdown feature on boards like the heltec v3? Meshtastic does *something* when you trigger the shutdown feature but I'm not sure if it's a true shutdown or just a really low power mode. Either way it would be nice to not have to have the node running constantly for battery-powered nodes

ripplebiz, 2025-04-13:
The Heltec boards are very basic, and don't have 'soft' power on/off, and for some reason a bunch of popular cases out there didn't think to include a physical on/off switch, instead relying on the Deep-sleep feature of Meshtastic, which is NOT off. It's just 'low power'.
A physical switch solves this issue.

closed 2025-04-13 by ripplebiz

neutralinsomniac, 2025-04-15:
not usually my style to comment on a closed issue so apologies, but the difference in power consumption between an all-on heltec and a deep-sleep heltec is significant enough to be a worthwhile feature imo despite it not being truly off. Requiring a physical power switch to stop your battery from draining in a day or less whether or not the node is being used locks you out of a ton of amazing switch-free cases (such as the ones by Alley Cat). If I (or someone else) puts forth the effort to implement this and PR it, is there any chance of it being accepted or would it be wasted effort? Thank you :)

recrof, 2025-04-15:
@neutralinsomniac I don't think it will be rejected, if you know what you're doing. I tried implement really basic deep sleep for v3 and it worked, but it was really crude and I did not implement any other wakeup interrupts apart from button itself.