meshcore-dev/MeshCore #169: [Feature Request] blacklist usage control for routers, rooms and client.

shadowdemon9956, 2025-04-01:
Andy Kirby mentioned the ability to control who can use repeaters in his first introduction video for Meshcore.  Could you please allow repeater and room owners the ability to block other nodes and repeaters from using or routing via our equipment . Meshtastic has a lot of bad actors with abusive node names and abusive messages, it will not be long before all of these parasites move over to Meshcore. A way to block access to these users would be very positive for MC.

seagull9000, 2025-04-30:
Perhaps the ability to upvote/downvote nodes, like with Survivor.
- A node starts life with a +ve number of ticks just for being put up.
- Other mesh nodes can upvote or down vote based on position/contribution to routing, appearance, behaviour.
- Select what threshhold of ticks get shown or used for routing if a repeater.
- Select what threshhold of ticks get acknowledged or messages displayed if a companion app.
- Select what threhhold of ticks get shown on map if mapping app.
- To reset ones rep, a reinstall would be required to get a new key.

Perhaps since the usefulness of a router is subjectiveto ones own position, these are separate ticks maintained on own node.

Just some thoughts.  In practice this is probably code/functionality that bloats the firmware somewhat.

I think this may hep address https://github.com/ripplebiz/MeshCore/issues/97 also.

liamcottle, 2025-05-01:
> @seagull9000 Perhaps the ability to upvote/downvote nodes, like with Survivor.

It's a fun idea! But I can totally see someone generating 10,000 unique identities and forcing the votes in favour of whatever they desire. Then the fun part is who maintains that database, and how is it kept in sync across all nodes on the network ;)

There's some work going on in the background for a Transport Codes system. This would allow for locking down the repeaters to specific groups of people that know the keys to use the repeaters. These keys would likely be changeable via remote management. If you find someone with access is being bad, you could rotate your keys to kick them out.

Still heavy work in progress, but network access and control of your hardware is most certainly a priority feature.

PE5TT, 2025-11-22:
No need to make it complex with voting. Just the ability to blacklist a client to stop spamming the network. As a repeater owner i would like control to who and what i repeat. Now that Meshcore gets extremely fast popular so is spamming. Mostly from a few users.

ase1590, 2026-08-29:
> As long as it’s possible to generate a new public key in a matter of seconds, I’m not sure a technical solution to this problem—which is obviously real—can be found. 

I don't think the solution needs to be bullet proof. 

Just a simple repeater specific deny. Sure they can bypass it but that also means spending effort on their part. It's like house locks on doors, which can just be picked in 5 seconds. Blacklists just make it slightly annoying enough for bad actors that the payoff isn't worth it.

NivvyDaSkrl, 2026-09-01:
> Creating blacklists will inevitably lead to splits in the mesh and the development of competing meshes that block each other; that’s always how it goes.
> 
> I also believe that the concept of a shared blacklist runs counter to decentralization and results in a group of users seizing control. This is not desirable.

Our mesh has some bad actors who cause intentional and malicious disruption to the mesh community in my city with botspam, forcing us to cave into whatever they want us to do in order for them to turn the bots off and let the rest of us actually use the network we built.

Blacklisting their repeaters would make it more difficult to circumvent blocking; right now they just re-key their companions. 

A community should be allowed to censure individuals who intentionally abuse it, and I as an individual should have the right to prevent the hardware I built and own from being abused to cause harm to my community.

jranma, 2026-09-01:
> Our mesh has some bad actors who cause intentional and malicious disruption to the mesh community in my city with botspam, forcing us to cave into whatever they want us to do in order for them to [...]

I agree with your reasoning, but I don’t understand how you plan to go about blacklisting a malicious actor. 
Blacklisting based on a nickname or public key number has virtually no effect at least on a determined user. 
Blocking based on a hardware identifier is not possible given the protocol’s design and/or the hardware’s design. 

In my opinion, the only way to do this is to implement limits on the number of messages sent per hour or minute from a public key. Hard-coded limits are needed to prevent the spread of excessive messages (configurable limits will be bypassed). But this would be a measure with limited impact, since such a limit would necessarily have to be high if we want to maintain a satisfactory user experience.

And even then, we’re still talking about the software layer. A malicious user could saturate the radio frequency and prevent other users from using the service, even though, in principle, the problem would remain localized here and the user wouldn’t benefit from the amplification effect that a software layer would entail.

PE5TT, 2026-09-01:
Please stop think that the only solution is a 100% state hacker proof implementation of banning . The bad actors are stupid lazy idiots that will get tired quickly when they get constantly banned. They are not hacker just assholes

enigmaspb, 2026-09-01:
I read all the discussions above

It seems to me that the repeater should be able to make 2 lists of allowed and prohibited nodes, separately for relaying and separately for direct.
This will solve quite a lot of problems, firstly, you can remove nodes that spoil (extend) the route from the list of prohibited for direct, and secondly, if necessary, you can make node repeaters only for inter-repeater communication using the list of allowed ones.

I see that there are often problems with the misuse of the meshkor network, for example, in Moscow, someone uses the meshkor network to transmit meter data from residential buildings, which creates a large load on the network, and many repeater owners would close such retransmission.

I see that there is a PR https://github.com/meshcore-dev/MeshCore/pull/2343 , but it is not updated for the current firmware version, and I do not have enough knowledge even with chatgpt to optimize it for 17.1

jranma, 2026-09-01:
Interesting discussion. I’m also looking for ways to prevent a few individuals from abusing the infrastructure and misusing shared resources. 

Leaving the technical aspects aside, let’s imagine that a blacklist system were added to the firmware. 

User X will immediately block user Y because Y keeps sending silly messages like “good night” or “received with 3 hops.”  In retaliation, Y will convince his group of friends to block the repeater belonging to X. To ensure that messages cannot be routed, Y will also block the repeaters belonging to Z, A, B, and C, which are located near X and have been chatting amicably with him. A group of relay administrators challenges this decision and considers it unfair: they decide, to set an example, to block Y.

Let’s let this situation play out for a few days, weeks, or months: we’re sure to end up with three networks that are blocking each other and seeking to increase their influence by capturing as many nodes as possible. 

Am I being too pessimistic, or can we count on users to use common sense?

PE5TT, 2026-09-01:
_"Am I being too pessimistic, or can we count on users to use common sense?"_

Your are beeing to perssimistic , 99 % of the people in the world are fine social beings. it is just this the 1% of  assholes ruining it for all of us.

NivvyDaSkrl, 2026-09-02:
> I agree with your reasoning, but I don’t understand how you plan to go about blacklisting a malicious actor. Blacklisting based on a nickname or public key number has virtually no effect at least on [...]

Antisocial people have less investment in breaking things than social people have in building and maintaining them. But the builders need the tools to do the maintenance.

> User X will immediately block user Y because Y keeps sending silly messages like “good night” or “received with 3 hops.” In retaliation, Y will convince his group of friends to block the repeater [...]

Yeah, that could happen.

> Let’s let this situation play out for a few days, weeks, or months: we’re sure to end up with three networks that are blocking each other and seeking to increase their influence by capturing as many nodes as possible.

I don't see an issue with multiple networks for different social groups. Reduced routing will ensure that even if the same frequency is used for all the meshes, the actual load on the frequency will be less. If it becomes a problem, it's likely one of the meshes will change frequencies.

> Am I being too pessimistic, or can we count on users to use common sense?

I think the main difference between our viewpoints is that I don't think a single unified mesh should be considered the ideal state that every metro area should strive for. I think communication technology should serve the people that use it, and if two communities decide they need to partition themselves, that should be easy to do without having to coordinate a LoRa settings migration.

Talie5in, 2026-09-02:
> The bad actors are stupid lazy idiots that will get tired quickly when they get constantly banned.

I hear what you are saying, but we still need to refer to recrof's comment

> you can't filter by public key - only time it's present in the actual traffic is when advert is sent.

When a companion sends a GRP_TXT (Channel Message), their source "Public Key" is not actually in the packet at all, so it's not actually filterable.

You could "filter" the repeater they are going through, if there is one in front of yours, but then your filtering all that traffic via that repeater, not the specific person/companion your targetting.

But then, they just move enough distance or re-point their antenna etc to go through another repeater, or they may already be visible through multiple repeaters.

You could filter the channel hash, if they are sending to a specific channel, of which there are only 255 hashes across the board, and it's easy enough for them to move to another channel and keep sending spammy data, eventually you'd have to block all channels, then what?

You can chain some filtering to try and achieve it, eg, comes via repeater with this 1/2/3 byte hash, going to this channel hash - but again, your likely to filter out others your not targeting, and again, if they just come via another repeater, your doing the same thing again.

There is no real easy solution to this problem.

Jizzo25, 2026-09-08:
Following up on my initial comment above (Nov 25) I can state that eventually we took matters in our own hands.

Over here, mesh performance is heavily degrading over the (non-)use of regions/scopes and different opinions on what the mesh is for.

This goes for local vs long-range, observing and publishing other user's communication (habits), internet connection, ping bots and much more.

All of these controversies amplified by the ever increasing user count.

While it may be controversial to some, we realized at an early stage that a "global" mesh would eventually become unusuable thanks to user behaviour. Just as Meshtastic did.

So, first, we reverted to focussing on reducing max hops and blacklisting bad actors, repeaters, non-regional paths and channels in order to maintain at least regional mesh performance.

This actually worked for some time and was greeted by the users.

As the mesh got more and more flooded with countless Home Assistant requests, we started filtering packet types which helped massively in keeping communication going on the repeaters.

However, that was out of the window when a growing number of ham radio operators finally caught on MC in 2026.

As they technically bypassed the existing, regulated infrastructure they also imposed on the mesh what they knew best from their world. Which, of course, is not helpful to a mesh in all aspects. At times it seemed many of them found their "Wild West", freed from the shackles of the regulated and protocolled ham standards. Some of them (and others) simply do not grasp the way a mesh works to this day.

For example, relaying traffic to a different MC "channel" does not equal relaying traffic to another frequency. The very same frequency still gets clogged with the spam of scripts.

This eventually led us to introduce whitelisting.

Granted, a last ressort which we did not choose lightly. We do advocate communication through MC since the very beginning that is why we started to cut out more and more of what is severely sabotaging the comms idea.

Back in November 25, I stressed that we do not want/see the need for closed repeater circuits. This has changed in the face of overwhelming comms-threatening spam on a technical level and ingnorance/complacency/hedonism on the human part.

Now, we of course cannot manage the RX side of things. This unfortunately is the nature of radio. However, by means of filtering and blocking we managed to carve out the core(!) of comms on the TX side.

As a side effect, our repeaters become dead ends to any packet that is not comms related which is also benefitial to those not being part of the repeater group. Our repeaters no longer serve as amplifiers for all the reqs, resps, anons, local gas prices, water levels, private solar arrays, door bells or whatever people find to be worth spreading through the entire mesh.

This notion of "self-defence" may be a controversial approach to some.

However, in our particular setting, striving for a "mesh for everyone" turned into striving for a "mesh for none" before we enacted counter measures.

What the mesh is - and how well it will work - is defined by its users and their behaviour. Technology is just a tool.

We have learnt this with Meshtastic the hard way. And this simply is history repeating with a way better system. The starter of this thread realized this already, too, when MC was not around as it is today.

People are divers in their wants and needs, their personal impulse control and sense for community. Hence, to us, the future of high-performing meshes is based on like-minded people behind the repeaters connecting locally, regionally and possibly beyond.

Technologically, this is backed by Meshcore with a vast array of filtering and blocking add-ons and "secret" regions to actually manage traffic and uphold comms in a dense cloud of spam.

Again, this approach may appear "antisocial" and "elite" to some. That we can't help as preserving reliable comms is our first and foremost priority.

We can only say that this is a reaction to what has been forced upon us. Mainly by those who came late.

Our way has proven fruitful to this day and this will conclude this real world report.

nullrouten0, 2026-09-08:
> So, first, we reverted to focussing on reducing max hops and blacklisting bad actors, repeaters, non-regional paths and channels in order to maintain at least regional mesh performance.

How?  This whole thread is about how this can’t be done.  Please enlighten us !

> As the mesh got more and more flooded with countless Home Assistant requests, we started filtering packet types which helped massively with keeping communication going on the repeaters.

Custom firmware?    How does it stop hundreds of non custom fw users in the same area?  

> As they technically bypassed the existing, regulated infrastructure 

How?

Thank you

enigmaspb, 2026-09-08:
> How? This whole thread is about how this can’t be done. Please enlighten us !

I'll guess the use of this https://github.com/meshcore-dev/MeshCore/pull/2343
block of spammer core repiter and spammer/flooding channels and blocking the path that includes these spammers repeaters

> Custom firmware? How does it stop hundreds of non custom fw users in the same area?

I'll guess
This is the firmware for the repeater with lists of blocked paths and repeaters ( https://github.com/meshcore-dev/MeshCore/pull/2343 ) , it does not affect users who do not use blocked repeaters

Jizzo25, 2026-09-09:
Given that a large portion of MC users is part of the problem rather than part of the solution (also on Github), we are, of course, well aware that it would not be very smart to elaborate in detail.

We have absolutely no love for MQTT, HA, bots and observers on Meshcore which limits our fan base by design.

It should have become clear that our dedication is fueled by defending reliable mesh comms against the recurring inconveniences of mass behaviour and misled individualism.

Hence, the layout of our journey.

All of the above (and more) can be done and has been done to great effect with Meshcore by skilled people - even without AI.

This will be/stay not exclusive to us.

But we are certainly not offering critical insights to parties interested in further degrading any mesh intentionally/unintentionally for whatever reason.

NivvyDaSkrl, 2026-09-11:
> Given that a large portion of MC users is part of the problem rather than part of the solution (also on Github), we are, of course, well aware that it would not be very smart to elaborate in [...]

If you believe that publishing a solution will cause the solution to be ineffective, then I'm not sure why you're commenting on an issue which is trying to get an open-source implementation of a solution.

msmuenchen, 2026-11-15:
still relevant

starvald, 2026-11-15:
risk of turning into a reference source and people using the 3rd party filter firmware's.

perhaps understandable, if it's 'baked in' then work arounds from spammers will more likely happen.