meshcore-dev/MeshCore #174: GPS for non GPS devices?

Emmpunkt, 2025-04-04:
Are there future plans to connect a GPS to non GPS devices like the Heltec V3 or similar?
Or is it even possible right now (and i couldnt find it).
Regards M.

liamcottle, 2025-05-24:
> Please add GNSS info from smartphone GPS

Maybe in a future update. But this isn't something I'm looking to do right now.

For the meantime, you can long press on the map to update your position, then send an advert for it to propagate to other devices on the mesh.

jdkruzr, 2026-02-17:
if this isn't the right repo to put this request in I'm happy to open another issue so this one can be closed, but this is really badly needed. what is the increased compliance burden for adding location? can we assist with this somehow?

emaxgms, 2026-08-25:
Hi everyone, is there any update on this ? 
It really is something very important to add.

hamnerd, 2026-10-16:
I know this has been on the logs now for 2 years. There are lots of devices that do not have GPS built in and frankly why buy an additional GPS when the phone has one built in. It would be great to have this feature available to make location updates easier for those who want their location known. Currently you have to scroll around the map to figure out where you are.