meshcore-dev/MeshCore #175: T-Echo battery life

akrabu, 2025-04-04:
I have recently switched my T-Echo from Meshtastic to MeshCore.  I have it in my truck, using the internal battery, coupled with a 20,000 mAh battery pack, plugged into my cigarette lighter and charging whenever I drive (which isn't often - just on the weekends, really).  Typically I could get about a month or more before needing to take the battery inside and charge it.

Since flashing to MeshCore (repeater role), I am now getting about a week at most.  Last night I took it off the battery, flashed it from 1.4.1 to 1.4.2, charged it up to 100%, then let it run by itself.  It managed 13 hours before it was dead.  There was about 18 seconds of airtime that entire time (so it's not busy in my area at all).  I'd typically get 4-5 days of battery life using Meshtastic, with just the internal battery.  Obviously something is burning the battery, but I'm not sure what.

I would assume I'd get better or at least the same battery life, since (a) there's no MC traffic in my area, (b) it's not using the screen, (c) MC transmissions are quicker and use less airtime anyway.  So I'm assuming there's a bug, and perhaps the T-Echo's CPU is stuck at 100% or something?  The only time the LEDs come on is when it's charging, so I don't think it's that.  Could be the GPS perhaps?

frantsch, 2025-04-14:
I can confirm the observation of x 1:1. I'm really looking forward to your corrected firmware version, which may provide a major remedy to the energy consumption of t-echoes.

frantsch, 2025-04-22:
the recent firmware v1.5.1 brought significant remedy turning off gps. niw a full battery seems to last for 48 hours roundabout.
many thanks for this solution!

recrof, 2025-04-22:
@akrabu if you're happy with battery longetivity, can you close this issue?

closed 2025-04-22 by akrabu