meshcore-dev/MeshCore #179: Unable to return to meshtastic firmware

lukaszraczylo, 2025-04-07:
Hi,
After flashing meshcore firmware on T1000-e, playing around I have decided to move back to the default meshtastic firmware. Unfortunately I am unable to do so.

1. "Custom" firmware from the flasher - timeouts
2. Flasher -> T1000e -> enter DFU -> back -> custom firmware - timeouts
3. `meshtastic` python client fails as well with timeouts.

Currently, I can only re-flash the meshcore firmware, which is far from optimal.
Any advice would be greatly appreciated.

lukaszraczylo, 2025-04-07:
Perfect, that sorted the issue.

Ps. I hope to come back to the meshcore as a contributor when a few more devices arrive ( most likely within next few days ) :)

recrof, 2025-04-07:
> Perfect, that sorted the issue.
> 
> Ps. I hope to come back to the meshcore as a contributor when a few more devices arrive ( most likely within next few days ) :)

we're looking forward to it!