meshcore-dev/MeshCore #191: t-echo: button functions

frantsch, 2025-04-11:
Just wondering if any functions (such as shutdown/start, reboot, etc.) are assigned to the two lilygo t-echo buttons?

recrof, 2025-04-11:
first "reset" button is hardware - you cannot change it. second one will be used for UI, which is not implemeted on t-echo yet. but it's planned soon.

frantsch, 2025-04-13:
I have a follow-uo question: 
Is there a firmware, or hardware way to shutdown a lilygo t-echo?

recrof, 2025-04-13:
it does not have ability to be shut down, you could put it into deep sleep tho. MeshCore currently does not have that implemented yet.

frantsch, 2025-04-13:
Slleping mode is what i meant. This is needen e.g. for changing antennas...

recrof, 2025-04-13:
you won't damage the sx1262 for few seconds needed to change the antenna

frantsch, 2025-04-13:
good to know, thanks!

AndreaCCIE, 2025-05-02:
Thanks @recrof, is there a road map for MeshCore?
I need to be able to deep sleep my nodes which don’t have physical power button.

recrof, 2025-05-02:
@AndreaCCIE we don't have roadmap currently. but you might be interested in this PR: https://github.com/ripplebiz/MeshCore/pull/226

446564, 2025-06-20:
can this be closed now?

JudiciousUneasiness, 2025-07-06:
Am I understanding correctly that there is not a way to shut off a TEcho with this firmware?  I just installed 1.7.2

JudiciousUneasiness, 2025-07-06:
> > Am I understanding correctly that there is not a way to shut off a TEcho with this firmware? I just installed 1.7.2
> 
> no, there is no deep sleep implemented. you can't really shut the node down without physical switch or dedicated power chip.

Thank you for clarifying that.  I have wanted to like the TEcho for its small form factor and eink but not having a power switch makes it pretty useless for me.  Even with the other guys' firmware, you can put it to sleep from the UI but if you so much as breathe on the user button it wakes back up, so it cannot ride in a pocket or backpack if you don't want it switched on.  My TEcho will probably be added to the sell/donate pile.

446564, 2025-07-06:
Also check out the nano G2 ultra

closed 2025-08-06 by recrof