meshcore-dev/MeshCore #201: Support for Beitian BN-180 GPS Module on LilyGo T-Deck Plus

foxhnd-git, 2025-04-13:
Hello,

I've flashed Meshcore 6.6 to my LilyGo T-Deck Plus devices, but the onboard GPS module is a Beitian BN-180, not the expected u-blox M10Q or Quectel L76K.

I'm having trouble acquiring a GPS fix, which appears to be due to an incompatibility with Meshcore. I've tried adjusting the baud rate to both 9600 and 38400, but neither worked. That said, the same module functions properly with Meshtastic and ChatterBox, so it seems to be a software-side issue rather than hardware.

I’d love to continue using and supporting Ripple Radio / Meshcore, and I’m hoping support for the BN-180 module can be added in a future update.

Here’s the [product page] (https://store.beitian.com/products/beitian-ubx-m8030-g-mouse-supports-gps-qzss-and-sbas-fixed-wing-traversing-aircraft-gps-module-antenna-bn-180-220-280-357-880-880q?variant=46694929957151) and [BN-180 GPS Module Datasheet.pdf](https://github.com/user-attachments/files/19727796/BN-180.GPS.Module.Datasheet.pdf) for your reference.

Thanks again for an awesome project!

ripplebiz, 2025-04-14:
The first GPS I used with the T-Deck was the BN-180, so it definitely supports it. It usually is just a 9600 baud setting needed.
In the GPS Info screen, does the 'Sentences' counter increment?

foxhnd-git, 2025-04-14:
> The first GPS I used with the T-Deck was the BN-180, so it definitely supports it. It usually is just a 9600 baud setting needed. In the GPS Info screen, does the 'Sentences' counter increment?

No, the 'Sentences' counter does not increment. I've tested both 9600 and 38400 baud rates. I'm located on the West Coast (USA), and I took both devices outdoors, waiting over 30 minutes with each baud setting, but still couldn't get a GPS lock.

LurkingFear, 2025-06-22:
Could your TX/RX wires be crossed?

foxhnd-git, 2025-06-23:
> Could your TX/RX wires be crossed?

I don’t think so, as it works fine with Meshtastic and ChatterBox, as I mentioned before. I’m using two devices, and both experience the same problem with MeshCore.

liamcottle, 2025-08-31:
It seems that `BN-180` is already supported. Will close this one out. Feel free to follow up if you're still having issues.

closed 2025-08-31 by liamcottle