meshcore-dev/MeshCore #203: XIAO ESP32S3 & Wio-SX1262 Kit Battery Voltage

Emmpunkt, 2025-04-14:
How can i get the Battery Voltage fom this device?
I guess there is no `PIN_VBAT_READ` defined, because the device has no build-in pin for this.
Most people use a voltage divider on pin A0 (D0), but how can i define it in the program?
I want to build a tiny repeater with it....
Regards M.

recrof, 2025-04-22:
@fdlamotte do you have any clue?

fdlamotte, 2025-04-22:
You can define your target with a PIN_VBAT_READ (it is used on the xiao_c3 port) 

I think it assumes a 1/2 ratio from the divider bridge (because that's what scott uses in his tuto), but we could make this adjustable.

I think it would be safe to have it enabled in default build. We'd have to make sure it is not used for other purpose with the official expansion board for seeed ...

Emmpunkt, 2025-04-22:
I´ve already compiled my own version, but its not very user friendly, not everyone is able to do it.
And you have to compile/change it with every new version of the firmware.....

recrof, 2025-04-22:
please open a PR, so we can incorporate it to firmware for everyone!

Emmpunkt, 2025-04-22:
@recrof Sorry, but i don´t know how this works, i´ve never done this before and i´m not a programmer ;)
Do i have to upload code?

recrof, 2025-04-22:
@Emmpunkt you just press fork button on the MeshCore, project, it will create copy of the repository in your account, where you can do all the changes, if you commit it, you can sync the changes back to original repo.

recrof, 2025-04-22:
that's called Pull Request(PR)

Emmpunkt, 2025-04-23:
I hope i did this right....

fdlamotte, 2025-04-23:
If not you'll learn ;)

liamcottle, 2025-08-31:
Closing this one out as it looks like PR #221 was rejected, due to the board not supporting this without a hardware modification.