meshcore-dev/MeshCore #209: Node Meshcore DIY

giobbe70, 2025-04-20:
hello 
I would like to create a Meshcore node. I took a 30 pin ESP32, a Lora SX1278 board and a mini oled display.
Which version of Meshcore should I flash? 
Thanks

recrof, 2025-04-20:
depends on the pinout. if it's different than other sx1278 nodes(tbeam, heltec v2), you will need to create new variant manually and flash it from platformio.

giobbe70, 2025-04-20:
my pinout

[Cd CWSA-with-magisk-GApps-pico_2204.txt](https://github.com/user-attachments/files/19824648/Cd.CWSA-with-magisk-GApps-pico_2204.txt)

giobbe70, 2025-04-20:
What is the pinout of the Heltec V2 to be able to duplicate it?
Thanks

recrof, 2025-04-20:
look at the `variants` folder of this project.

giobbe70, 2025-04-20:
I am a beginner, please can you create a version for my hardware ? 
So that I can flash it easily.
Thanks

recrof, 2025-04-20:
I'm affraid you'll need to learn a bit if you want to make custom firmware for your dyi hardware. it's not hard to do a basic port.

giobbe70, 2025-04-20:
Meshtastic have a DIY version, work very well....
Because Meshcore haven't a DIY version???

recrof, 2025-04-20:
here is quick guide how to add new variant and flash it to your device:
1. git clone this repo
2. install vscode and platform.io extension
3. open the MeshCore folder in vscode
4. let platform.io download all the required libs
5. duplicate heltec v2 under variants folder and edit platformio.ini - you need to update names, identifiers and all pin numbers. you probably don't need to touch `board` tho.
6. save and restart vscode, newly created roles for your variant should be ready to be built and flashed
7. choose your platform.io variant/role from bottom panel, connect it to your computer and flash it by selecting platform.io icon from left bar(little alien icon) and you chose "upload" after everything updates - this will flash the board.

giobbe70, 2025-04-20:
In my Lora SX1278 board have pin MISO and MOSI and SCK. Heltec V2 have pin SDA and SCL.

recrof, 2025-04-20:
SDA and SCL are display data pins. you don't need to define them if you don't need working display. miso & mosi pins definition can be seen used here: https://github.com/ripplebiz/MeshCore/blob/main/variants/lilygo_tlora_v2_1/platformio.ini

giobbe70, 2025-04-22:
No,

can you send me a custom version of Meshcore for my board please

Thanks

recrof, 2025-04-25:
@holoGDM look at definitions for tlora 1.6 - all the required pins are in platformio.ini. but I don't think you need to actually create new device variant to make it work.. just flash WSL3 firmware and it should spring to life(without gps). we will try to create porting guide, for now, please see https://github.com/ripplebiz/MeshCore/pull/222/commits/da1febdd88a1d46abce859d3f6710f1b87295572 as reference how to create new variant. there are plenty other examples in closed pull requests.

giobbe70, 2025-04-25:
Do Meshcore work with card Lora SX1278?
Thanks

seagull9000, 2025-04-30:
I hope you can help with this one - I've been  trying for three days to get the TFT display to work for the Heltec Wireless Tracker.

I used the WSL platformio.ini and added the following:

```
[env:Heltec_Wireless_Tracker_companion_radio_ble]
extends = Heltec_lora32_v3
build_flags =
  ${Heltec_lora32_v3.build_flags}
  -I src/helpers/ui
  -D ST7735
  -D DISPLAY_CLASS=ST7735Display
  -D USE_PIN_TFT=1
  -D MAX_CONTACTS=100
  -D MAX_GROUP_CHANNELS=8
  -D BLE_PIN_CODE=735735
  -D BLE_DEBUG_LOGGING=1
  -D P_LORA_TX_LED=21
  -D PIN_TFT_SDA=42
  -D PIN_TFT_SCL=41
  -D PIN_TFT_DC=40    # RS (register select)
  -D PIN_TFT_RST=39   # RES 
  -D PIN_TFT_CS=38
  -D PIN_TFT_VDD_CTL=3  
  -D PIN_TFT_LEDA_CTL=18
;  -D ENABLE_PRIVATE_KEY_IMPORT=1
;  -D ENABLE_PRIVATE_KEY_EXPORT=1
;  -D MESH_PACKET_LOGGING=1
;  -D MESH_DEBUG=1
build_src_filter = ${Heltec_lora32_v3.build_src_filter}
  +<helpers/ui/ST7735Display.cpp>
  +<helpers/esp32/*.cpp>
  +<../examples/companion_radio>
lib_deps =
  ${Heltec_lora32_v3.lib_deps}
  densaugeo/base64 @ ~1.4.0
  adafruit/Adafruit ST7735 and ST7789 Library @ ^1.11.0
```

But the TFT display never shows up.   I have tried it with the ST7789 ui class as is, plus I copied the ST7789 to ST7735 in case that made a difference.  It didn't.

I tried putting in some Serial.println statements for debugging, but they never show up in the terminal.  I could probably make more progress if I could get some debugging output.

Any thoughts?

recrof, 2025-04-30:
it's not that easy.. apart from importing the display library, you need to create wrapper class to initialize it and drive it. look at `src/helpers/ui` directory where all the display classes are defined. UILib might also need some tuning for this display, as it's way smaller than SSD1306.

seagull9000, 2025-04-30:
I thought the ui wrappers were done.  I'll persevere.

Is there a trick to getting debug statements to work?

recrof, 2025-04-30:
did you send the serial debug output after serial init was done? that's the most common problem.

seagull9000, 2025-04-30:
> did you send the serial debug output after serial init was done? that's the most common problem.

I put a test statement in main.cpp.  Here's the code fragment:

```
void setup() {
  Serial.begin(115200);
  Serial.println("Companion Wireless Tracker - started");

  board.begin();
```

seagull9000, 2025-04-30:
I'll open a new issue for Heltec Wireless Tracker in case this helps anyone with getting it going.

closed 2025-08-31 by liamcottle