meshcore-dev/MeshCore #215: Rak user button

ch3p4ll3, 2025-04-21:
I flashed the latest firmware in Ble version for my rak wisblock board. Before meshcore I had flashed meshtastic and the button worked(as per meshtastic guide I used PIN 31 aka AIN1). I flashed the file in meshcore releases but upon pressing the button nothing happens, the display stays off. I also tried changing the button parameter by changing the PIN from 9 to 31, recompiled and loaded the new firmware but still nothing.

recrof, 2025-04-21:
the code that we use is tested only on wismesh RAK19026 board user button - this one needs `pinMode(PIN_USER_BTN, INPUT_PULLUP);` in `src/helpers/nrf52/RAK4631Board.cpp` - I would look in the RAK documentation if the same is needed for 19007 board user button. If not, detection of correct board is needed to check which button needs to be used and how to initialize it.

MC-DE-Wiki, 2025-04-25:
Same here with RAK 4630  ¯\\\_(ツ)_/¯

memo-567, 2025-04-25:
Same here with RAK 4630  ¯\\\_(ツ)_/¯

ch3p4ll3, 2025-04-25:
I tried setting pin 31 as `PIN_USR_BTN` , compiled and uploaded on my rak 19007 but still no response. `pinMode(PIN_USER_BTN, INPUT_PULLUP)` should work also con rak 19007 with rak 4630 as core module. 

Lastly I tried setting the `PIN_USR_BTN` to `A1` and that bricked the core module and I had to reflash the bootloader as well as the softdevice with and external programmer.

liamcottle, 2025-04-26:
Can confirm user button on WisMesh Pocket RAK19026 is broken on v1.5.1, downgrading to v1.4.3 it works again.

- v1.4.3 user button wakes the screen
- v1.5.1 user button does nothing

Possibly related to recent screen changes that added `_need_refresh`?

liamcottle, 2025-04-26:
It appears some code was added for the user button pin in the `UITask`.

https://github.com/ripplebiz/MeshCore/blob/76847a77561d376a31af5b0b2cad192f419e787c/examples/companion_radio/UITask.cpp#L47

This unfortunately overrides the `INPUT_PULLUP` set in `RAK4631Board.cpp`.

https://github.com/ripplebiz/MeshCore/blob/76847a77561d376a31af5b0b2cad192f419e787c/src/helpers/nrf52/RAK4631Board.cpp#L26

This will need to be removed from `UITask.cpp` and moved to a board specific class if required to be in `INPUT` mode for another board.

Tagging @fdlamotte since they made that commit ;)

liamcottle, 2025-04-26:
PR to resolve https://github.com/ripplebiz/MeshCore/pull/228

fdlamotte, 2025-04-26:
You're totally right, this shouldn't be here ... probably left after some tests :/

adam2872, 2025-05-12:
I've created a PR for this issue: https://github.com/ripplebiz/MeshCore/pull/280
It allows the use of the analogue pin 31 input on a RAK4631 board to be used as the user button - same as on MT

liamcottle, 2025-08-31:
Closing out as #228 and #416 were merged and fixed this issue.

closed 2025-08-31 by liamcottle