meshcore-dev/MeshCore #224: T1000-E high battery drain

neutralinsomniac, 2025-04-24:
I'm assuming this is another case where we should probably disable GPS. Normally my T1000-E lasts several days on a single charge, but currently I've had it running for ~2 hours and it's already drained 17% which means I'll be happy to get 10 hours out of it.

446564, 2025-04-24:
I've been using @fdlamotte experimental firmware and getting about 2 days. I'm sure it will make it's way upstream eventually. https://github.com/fdlamotte/MeshCoreT1000e

fdlamotte, 2025-04-24:
there seems to be something wrong on the main repo then ...
some optimization for bat have been upstreamed there, you should get 2 days normally, but I didn't test with last firmware
please test with custom firmware to see if problem doesn't happen there ...

neutralinsomniac, 2025-04-24:
battery seems to have settled down. currently at 81%. I'll report back once it hits sub-10%.

neutralinsomniac, 2025-04-25:
I'm at 63% overnight. gonna chalk this up to a weird initial battery reading. closing this issue :)

closed 2025-04-25 by neutralinsomniac