meshcore-dev/MeshCore #227: Quick start guide for new hardware

seagull9000, 2025-04-25:
I have cloned the meshcore repository into vscode and can compile one of the examples for a device successfully, but now I want to create a variant for a Heltec Wireless Tracker node. 

It works as a companion radio using the Heltec Wireless Stick firmware, but neither the oled nore gps is working.

I'm happy to build this support myself, but a quick start guide would be helpful.

I am also happy to contribute the variant but not sure how to do that.

Thanks

recrof, 2025-04-25:
Thanks for the report. Until we have new device onboarding guide, please see few of the pull requests that will tell you what needs to be done to add new device: https://github.com/ripplebiz/MeshCore/pull/222/commits/da1febdd88a1d46abce859d3f6710f1b87295572

seagull9000, 2025-04-26:
> Thanks for the report. Until we have new device onboarding guide, please see few of the pull requests that will tell you what needs to be done to add new device: [da1febd](https://github.com/ripplebiz/MeshCore/commit/da1febdd88a1d46abce859d3f6710f1b87295572)

Excellent , thanks.  I will give that a go..