meshcore-dev/MeshCore #239: Heltec Wireless Tracker build

seagull9000, 2025-04-30:
This is a continuation of the discussion in https://github.com/ripplebiz/MeshCore/issues/209

Progress so far:

If you flash the Heltec Wiress Tracker (HWT) with the companion radio ble firmware for the Heltec WSL (Wireless Stick), the HWT will work.  It participates in the mesh without any problems I coud see.

However, we want more.  The HWT has a TFT display.  The documentation on the Heltec site shows the display is a ST7735.

In the file:  variants/heltec_v3/platformio.ini, I copied the ```[env:Heltec_WSL3_companion_radio_ble]``` section into a new ```[env:Heltec_Wireless_Tracker_companion_radio_ble]``` section.

Using the schematic found here https://resource.heltec.cn/download/Wireless_Tracker/Wireless%20Tracker1.1.pdf I ascertained the pins to the display and the LED.

This yielded the following section in the platformio.ini file:

```
[env:Heltec_Wireless_Tracker_companion_radio_ble]
extends = Heltec_lora32_v3
build_flags =
  ${Heltec_lora32_v3.build_flags}
  -I src/helpers/ui
  -D ST7789
  -D DISPLAY_CLASS=ST7789Display
  -D USE_PIN_TFT=1
  -D MAX_CONTACTS=100
  -D MAX_GROUP_CHANNELS=8
  -D BLE_PIN_CODE=735735
  -D BLE_DEBUG_LOGGING=1
  -D P_LORA_TX_LED=21
  -D PIN_TFT_SDA=42
  -D PIN_TFT_SCL=41
  -D PIN_TFT_DC=40    # RS (register select)
  -D PIN_TFT_RST=39   # RES 
  -D PIN_TFT_CS=38
  -D PIN_TFT_VDD_CTL=3  
  -D PIN_TFT_LEDA_CTL=18
;  -D ENABLE_PRIVATE_KEY_IMPORT=1
;  -D ENABLE_PRIVATE_KEY_EXPORT=1
;  -D MESH_PACKET_LOGGING=1
;  -D MESH_DEBUG=1
build_src_filter = ${Heltec_lora32_v3.build_src_filter}
  +<helpers/ui/ST7789Display.cpp>
  +<helpers/esp32/*.cpp>
  +<../examples/companion_radio>
lib_deps =
  ${Heltec_lora32_v3.lib_deps}
  densaugeo/base64 @ ~1.4.0
  adafruit/Adafruit ST7735 and ST7789 Library @ ^1.11.0
```

The define
```  -D USE_PIN_TFT=1``` 
is required for getting the ST7789Display to initialise without needing SPI or SPI1.

```
#pragma once

#include "DisplayDriver.h"
#include <Wire.h>
#include <SPI.h>
#include <Adafruit_GFX.h>
#include <Adafruit_ST7789.h>

class ST7789Display : public DisplayDriver {
  Adafruit_ST7789 display;
  bool _isOn;
  uint16_t _color;

  bool i2c_probe(TwoWire& wire, uint8_t addr);
public:
#ifdef USE_PIN_TFT
  ST7789Display() : DisplayDriver(160, 80), display(PIN_TFT_CS, PIN_TFT_DC, PIN_TFT_SDA, PIN_TFT_SCL, PIN_TFT_RST) { _isOn = false; }
#else
  ST7789Display() : DisplayDriver(135, 240), display(&SPI1, PIN_TFT_CS, PIN_TFT_DC, PIN_TFT_RST) { _isOn = false; }
#endif
  bool begin();

  bool isOn() override { return _isOn; }
  void turnOn() override;
  void turnOff() override;
  void clear() override;
  void startFrame(Color bkg = DARK) override;
  void setTextSize(int sz) override;
  void setColor(Color c) override;
  void setCursor(int x, int y) override;
  void print(const char* str) override;
  void fillRect(int x, int y, int w, int h) override;
  void drawRect(int x, int y, int w, int h) override;
  void drawXbm(int x, int y, const uint8_t* bits, int w, int h) override;
  void endFrame() override;
};
```

Where this is at:
- no luck getting the display to come to life.
- The LED comes on and stays on until the user button is pressed

Any advice would be welcome.

seagull9000, 2025-05-01:
Update:
Got display to switch on.  The clue was in the schematic with LED_K  switching ground through a mosfet.

Here is the current platformio.ini:

```
[env:Heltec_Wireless_Tracker_companion_radio_ble]
extends = Heltec_lora32_v3
build_flags =
  ${Heltec_lora32_v3.build_flags}
  -I src/helpers/ui
  -D ARDUINO_USB_CDC_ON_BOOT=1
  -D ST7789
  -D DISPLAY_CLASS=ST7789Display
  -D USE_PIN_TFT=1
  -D MAX_CONTACTS=100
  -D MAX_GROUP_CHANNELS=8
  -D BLE_PIN_CODE=735735
  -D BLE_DEBUG_LOGGING=1
  -D P_LORA_TX_LED=18
  -D PIN_TFT_SDA=42   # SDIN
  -D PIN_TFT_SCL=41   # SCLK
  -D PIN_TFT_DC=40    # RS (register select)
  -D PIN_TFT_RST=39   # RES 
  -D PIN_TFT_CS=38
  -D PIN_TFT_VDD_CTL=3       # connected to VDD which is also connected to LEDA
  -D PIN_TFT_LEDA_CTL=21     # LEDK (switches on/off via mosfet to create the ground)
;  -D ENABLE_PRIVATE_KEY_IMPORT=1
;  -D ENABLE_PRIVATE_KEY_EXPORT=1
;  -D MESH_PACKET_LOGGING=1
;  -D MESH_DEBUG=1
build_src_filter = ${Heltec_lora32_v3.build_src_filter}
  +<helpers/esp32/*.cpp>
  +<../examples/companion_radio>
  +<helpers/ui/ST7789Display.cpp>
lib_deps =
  ${Heltec_lora32_v3.lib_deps}
  densaugeo/base64 @ ~1.4.0
  adafruit/Adafruit ST7735 and ST7789 Library @ ^1.11.0
```

The meshcore text come up but the screen needs to be rotated and there seems to be some offset at play.

seagull9000, 2025-05-01:
It lives!

The display (and the LED) is now working.  I'm not sure how to integrate the changes into the code base, but I'm happy to post the code here if someone is interested in doing it.  

Could one of the devs perhaps advise best course of action...

recrof, 2025-05-01:
the best course of action is to clone the repo on your github, commit the changes and then create a PR(pull request). https://github.blog/developer-skills/github/beginners-guide-to-github-creating-a-pull-request/

seagull9000, 2025-05-03:
> the best course of action is to clone the repo on your github, commit the changes and then create a PR(pull request). https://github.blog/developer-skills/github/beginners-guide-to-github-creating-a-pull-request/

Thanks recrof.  There are some outstanding things I need to nail down then I'll take a look at doing a pull request:
- When a message arrives it appears briefly on the display then almost immediately gets overrwritten when the display refreshes
- I haven't tested it with the simple repeater application
- The battery level isn't detected
- I've made no attempt to see if the GPS works yet

recrof, 2025-05-03:
@seagull9000 battery voltage routine should probably be in the factory firmware in heltec github repo. gps will probably need to use new sensor framework @ripplebiz created today.

jquatier, 2025-05-05:
I can help get the basics into a PR, if needed. I have one of these devices. GPS can probably come later.

seagull9000, 2025-05-06:
> I can help get the basics into a PR, if needed. I have one of these devices. GPS can probably come later.

Oh cool.  Thanks.  I just need to get back into it  once I have other priorities out of the way.  Power should be straight forward, I hope.

The main change to get the display working was to use the ST7735 driver library instead of the ST7789 that is already in the project, along with the pin assignments as above.  I've replicated the ST7789 files into ST7735 and made a small change in main.cpp for the board to pick up the right library.

seagull9000, 2025-05-06:
Battery level reporting is working now.

I tried to follow the pull request guide and have a token but when I try to push, I get the error:
```
remote: Permission to ripplebiz/MeshCore.git denied to seagull9000.
fatal: unable to access 'https://github.com/ripplebiz/MeshCore.git/': The requested URL returned error: 403
```
I guess just anyone can't push to the repository, which would make sense ...
Where do I go from here?

recrof, 2025-05-06:
you need to clone it to your github account first(clone button on the ripplebiz/MeshCore repo), then git clone it locally, then edit the code, git add/git commit/git push and then you can finally create pull request in your github clone repo. don't forget to clone the "dev" branch - you'll need to also create the pull request for "dev" branch as well.

seagull9000, 2025-05-07:
A pull request has been created for this:  https://github.com/ripplebiz/MeshCore/pull/258

This is my first PR - let me know if anything needs to be changed.

seagull9000, 2025-06-30:
Initial implementation completed

closed 2025-06-30 by seagull9000