meshcore-dev/MeshCore #259: Security issues in encryption!

trinami, 2025-05-07:
Dear MeshCore Team,

We have identified several security issues in the AES encryption implementation used in your project, including the use of ECB mode, which can expose the system to pattern attacks, and an inadequate padding method that may leak plaintext length information.

We would like to propose a solution to these vulnerabilities and can submit a pull request on GitHub to implement a more secure encryption mode and proper padding scheme in the coming days.

Best regards,

Trinami :)

EternityForest, 2025-05-09:
Looks like there's also not much replay attack prevention, seems like that should be done at a lower level so every packet has at least some degree of it.

ripplebiz, 2025-05-09:
There's quite a lot of replay attack guards, but it's at a higher level. It's not at the packet level. To do that would require extra data in headers, which is a waste when it can be done after payload is decrypted.

shadowdemon9956, 2025-05-09:
I'm not sure about the rest of the users here but I don't really care if my message asking Dave what antenna or solar panel he is using on his nodes gets decrypted and shared. The fact that Meshcore works well at long distance and works fast is more than enough for me. If I need to exchange delicate information to another person I will use a more secure method. The only weakness I could see is when logging into a remote repeater ,at worst it will just get hijacked, We would then simply exchange the node and carry on.
I don't think we need extra headers taking up bandwidth to secure our bland messages. Good spot though and no disrespect intended.

EternityForest, 2025-05-09:
I don't think the issue is so much about privacy, it's about sabotage and
DoS attacks.  If you want to control some lights or sprinklers or
something, someone's might come by and mess with them just to annoy you,
especially if any kind of internet connected stuff gets exposed later.

Plus, mesh tech works better when there's more users, so it's nice not to
have two separate technologies just because one or two things need to be
secured.

shadowdemon9956, 2025-05-10:
I use multiple technologies for different use cases, as we all do. Do you use a Lora mesh for mobile banking? 
I think you are misinterpreting the concept of this platform, this is a repeater based messaging system for people that have an interest in radio without the price tag of HAM radio transceivers, We are all in it for the fun and the achievements. 
And yes, I do remotely control systems within my home but I use GSM to do so, and anybody with the knowledge to setup such a system over a Lora network will know the risks.
If you need a platform to fix then maybe direct your concerns to the automotive industry where the the replay, relay and DOS attacks cause actual loss and risk to life.
Lastly, it is call Meshcore but I don't think it is a mesh by any means, its way better with the ability to direct packets via repeater nodes as needed and see what happens on the path, talking to Steve that's 70 miles away from a 150mW Lora chirp transceiver module that's half the size of a dollar coin on a predefined path.

trinami, 2025-05-10:
Is it even possible to adjust the encryption without further complications while remaining compatible? I believe it would require a significant change, potentially necessitating a kind of Package 2.0.

Regarding implementation: If data is sent every second (though I'm not sure how often you send), then an additional 3 bytes (24 bits) would suffice:

 (((2 ^ 24) / 60) / 60) / 24  ≈ 194.18 days

This means that the IV would repeat approximately every 97 days. For each additional bit, the duration would double.

From a performance standpoint, this should not impact encryption significantly, as it would only add an XOR and copy function. The challenge lies more in sending the additional 3 bytes.

Another concern is the random value used; it must be truly random. I'm not sure if you have already implemented something like this.

Unfortunately, we noticed that the part that interests us, (t-deck) is not open source, so we will not invest any time more into it.

Best regards.

EternityForest, 2025-05-10:
The really simple way would be to use a 5 byte timestamp, Unix time plus
256ths of a second, that way you can detect old replayed messages.
Authentication isn't nearly as useful if replay attacks exist.

Fully random values are pretty much trivial though, most modern chips
already have a HW RNG onboard.

EternityForest, 2025-05-18:
Looking further into it, seems like pretty much every encrypted packet is already authenticated, so you could fix the ECB issue by   moving to GCM or something, without much disruption if you did it in stages:

* Make the decrypt code try decryption as GCM is ECB fails, tell people to update or they won't be able to receive from new nodes in a year

* Once that is already out there and everyone has that update, make the encryption code use GCM, and to try GCM first and then ECB

* Get rid of ECB entirely

inversenebula, 2025-05-18:
As a user, properly-implemented encryption is a big selling point of Meshcore for me and quite a few others that I know of. So, in contrast to others here, even if I personally did not value it as much, I would not volunteer to sacrifice that feature because maybe I personally don't think I need it and I would hope others here will offer the same consideration to those of us who do value it. Once a developer has sold (literally in this case, re: those of us who have paid) their messaging as encrypted they are really obligated to make sure it's working properly after that, not just say "oh never mind we don't really need that" because a few users are willing to sacrifice it. Many of us are NOT willing to sacrifice it. And I would hope and expect the devs to not cave in to this type of sentiment when it runs counter to one of the big selling points of this type of messaging network.

ripplebiz, 2025-05-19:
> Looking further into it, seems like pretty much every encrypted packet is already authenticated, so you could fix the ECB issue by moving to GCM or something, without much disruption if you did it [...]

Yes, this is probably the least disruptive way to introduce it. Once things settle down a bit, I'll try introducing this and get an eager guinea pig group to give it a test.

bconway, 2025-11-05:
Is there a reason such a young project chose ECB in this decade? Maybe there is some rationale that isn't obvious (or in the FAQ) that would add to the conversation.

nextgens, 2026-01-29:
@knr1  I don't have a problem with asacon: I think it's great for what we want to do here: we need to minimise the costs of initialisation and asacon does that.

Be wary of benchmarks such as https://www.oryx-embedded.com/benchmark/espressif/crypto-esp32-s3.html : they show that hardware-assisted stuff (AES typically) beats hand down any combination and ChaCha20Poly1305 is the second best choice. This is deceptive: we care about working on payloads <256bytes in size, not peak performance with larger payload sizes. I suspect @liquidraver on #1450 benched it wrong :) Our usecase ought to be "small payloads", "lots of different keys" (#1450 is barking at the wrong tree IMHO: the tag can be much shorter, you just need to vary keys).

A great writeup was recently posted on discord: https://jacksbrain.com/2026/01/a-hitchhiker-s-guide-to-meshcore-cryptography/
I'd add a few things: the 2 bytes (!?!) MAC is not verified in constant time, it is also keyed using the same key as the cipher ('cause why not /s). The keys themselves are never varied, their lifetime is unlimited...

IMHO, the real decision to be made is around how much of MAX_PAYLOAD (184 bytes) do we sacrifice to security? The current code uses a 2 bytes MAC; some of the encrypted payloads do not feature a 4 bytes timestamp. The current overhead is exactly 2 bytes.

For comparison, MT uses a 4 byte packet counter (https://github.com/meshtastic/firmware/issues/4031) and build a 31 bytes nonce out of that.

If we overall the whole packet structure I think we should look into having some "network consensus" (whether it's time #1443 , counter based or even blockchain-type is TBD) on something and use that in key derivation/nonce generation. If we do that most of it can be implicit (or shared on other packets) and the "wire costs" can be real-low while getting real security.

If on the other hand we hack something over the existing packet structure (like #1450) we are more limited. There my recommendation would be to (a) look at diversifying keys -whether using time- or just a PRF (HMAC(shared_key, counter|...) and get nonce and keys out of that) (b) truncate the tag: 2bytes was probably pushing it but if your keys change you only care about *active* attacks against integrity... the attacker is rate-limited by what LoRa lets him attempt until the key changes (c) look seriously into nonce: whether parts of the payload are reusable (think the 4bytes of timestamps) that could be transmitted in plaintext or hashed-in to help derive better nonces

@liquidraver if you're on discord let's chat :)

nextgens, 2026-01-29:
re-reading the above I need to clarify a few things:

1) short tag sizes are fine since the attacker is constrained by the bandwidth available; If he can send fast he will be able to forge stuff unless the tag is long enough (regardless of rekeying)
2) speaking with @liquidraver we don't need fancy for nonces, we could just have a boot_id (generated from the existing RNG at boot) concatenated with a counter. We need uniqueness out of nonces, nothing else.
3) There is no reason not to rekey often. Given that we currently do zero optimisation crypto wise (level one would be to safe the key schedule since we never rekey), we could even afford to have per message keys.

knr1, 2026-02-15:
OK, so I must have gotten the security 'vibes' wrong within this project:

When I first read the mission-goal...

> Our mission is to create the most reliable and secure decentralised mesh radio networking system which can be used for communications by anyone. Our focus is on security, reliability and efficiency.

... I really meant 'security' as of 'Snowden'-ready... (you know, as this guy seems to be part of the group 'anyone').
However, being allowed to be involved in this thread and while reading the really impressive blog indicated by @nextgens  (https://jacksbrain.com/2026/01/a-hitchhiker-s-guide-to-meshcore-cryptography/) I literally learned about another 'vibe' of security in this project: 

> Would I leak Snowden secrets? Hell no. Would I coordinate a festival meetup location or chat about my day? Absolutely.

Which is in line with @nextgens 
> IMHO, the real decision to be made is around how much of MAX_PAYLOAD (184 bytes) do we sacrifice to security? The current code uses a 2 bytes MAC; some of the encrypted payloads do not feature a 4 bytes timestamp. The current overhead is exactly 2 bytes.

This way, I can understand any sacrifice that are made for this project, but then one should maybe be straight about the true threat model, this project is able/willing to operate in, and be more explicit on who is meant by  'anyone' that should(/n't) use this way for communication?!

3dpgg, 2026-02-15:
I'm fairly new to Meshcore, but found this when looking through the cryptographic utilities, and that pointed me to this issue about using AES-ECB.

@RangerMauve , I think you made the same mistake that I did, which was locating the wrong cryptographic library. I too thought it was [this one](https://github.com/rweather/arduinolibs), but I believe it's [actually this one](https://github.com/OperatorFoundation/Crypto), since only the latter has a [version tag of 0.4.0](https://docs.arduino.cc/libraries/crypto/).

In any case, though, both copies have an [AES-CTR example](https://github.com/OperatorFoundation/Crypto/blob/master/examples/TestCTR/TestCTR.ino#L93), using a template for CTR of the form `CTR<AES128>`. The same form works for AES256, so there would not be an immediate need to transition to a new cryptographic library, at least not for this issue.

That said, I see that another contributor already opened PR #1677 for adding ChaCha20-Poly1305, and while a change from AES-ECB to AES-CTR would be a massive improvement, there will already be some incompatibility anyway. So might as well make a huge leap to a very modern cryptographic construction (with AEAD), with an eye to deprecate AES-ECB as quickly as possible.

It seems that that PR is also using the existing library's support for ChaCha20, so there shouldn't need to be any library upgrades in any case.

dreirund, 2026-02-23:
> *The really simple way would be to use a 5 byte timestamp, Unix time plus 256ths of a second, that way you can detect old replayed messages.*

Although I also object stating that "MeshCore has secure encryption", I argue against the timestamp detection for replay attacks:

In MeshCore devices, times tend to be wrong because devices might just restart and time is reset to some default in e.g. 2024. We do not have reliable means to set correct time, and we also cannot rely on that devices have a consistent clock at all.

3dpgg, 2026-04-13:
Whoops, looks like I wasn't clear enough. I was referencing your immediately-preceding reply, this portion: 

> If the frame format is not flexible enough (of which MeshCore v1 is certainly in this category), then, yes, that's a legitimate concern. I mean, that already happened and that's what this ticket is about.

But rereading it now, you don't explicitly mention version strings, although I gather that your point was that the existing header format is not extensible enough to accommodate a new crypto suite without a breaking change.

My apologies for the misquote. I wanted to underscore that the ideas I mentioned are better articulated in that blog post, which may also help describe how exactly a transition period would be managed at the protocol level.

darconeous, 2026-04-14:
We could go down a whole lot of rabbit holes from here. There are SO MANY THINGS I want to reply to in what both @knr1 and @3dpgg said, but I fear that we are just going to go down the rabbit hole again. And I think there is a deeper issue that needs to be worked out first.

I feel like the most pressing question is this: Who is the decision maker? Who has the authority to either make these architectural changes or approve the designs that others have made? That person (or persons) needs to chime in so that we can figure out what paths forward there are and what paths are non-starters. Otherwise we risk wasting a lot of effort.

I feel like what is needed is a designated steward to help shepherd these processes forward. Someone who helps gather consensus, split off discussions into their appropriate threads, keep track of progress, and determine when an issue has consensus so that everyone can move forward. In the IETF process, this person is the *working group chair*. We could call them "pizza lord" for all I care, but there really should be someone actively involved who is at least deputized to make decisions.

If there is already such a person for this task, I'm not sure who it is.

But without buy-in or direction from the project owners, I'm not sure what next steps we can or should take.

nextgens, 2026-04-15:
Consensus based decisions on crypto matters... Referring to ietf as a model... Is definitely not an approach I am comfortable with.

We need something that works quickly, not next century...

IMO what will end up being merged is the minimal set of changes that addresses current vulnerabilities.

We do not even share a common understanding of what these are... Usage of ECB is the tip of the iceberg.

darconeous, 2026-04-15:
For the sake of completeness, here are the benchmarks for AES-OTRv3 on ESP32-S3, NRF52, and my laptop (for reference):

https://gist.github.com/darconeous/2cc552690a9bf6f7ca4f2ca9de04c832

I don't have an STM32 board, so I couldn't test that platform. The repository for creating those benchmarks is [here](https://github.com/darconeous/aes-otr-embedded-benchmark) if anyone wants to try to reproduce. It was written with ChatGPT 5.4, which I gave @nextgen's prompt along with additional instructions to test the implementation first using known test vectors to validate the implementation.

nextgens, 2026-05-26:
So it's 4 bytes less than my proposal in https://github.com/meshcore-dev/MeshCore/issues/259#issuecomment-4172827299 ... and a lot less complex.

Whether it's secure or not, as you have identified, will depend on how you procure the IV. The current code sucks and is probably way too slow to be used as is (hence [#2280](https://github.com/meshcore-dev/MeshCore/pull/)). Granted you could have the same thing with AES-CTR though...

In the end, you end up with (very) weak integrity protection, you remain on static keys (used in both direction, re-used for the MAC, ...) so you do not get any forward security nor post-compromise security. You also get catastrophic failure modes if you fail to procure proper randomness.

I have not proposed anything with the current AES code because the current implementation sucks and I personally quite like the idea of [removing that library to save 49kB](https://github.com/meshcore-dev/MeshCore/pull/1632). On ESP32 we use [AESEsp32](https://github.com/rweather/arduinolibs/blob/master/libraries/Crypto/AESEsp32.cpp) and elsewhere I think we are on [AESSmall128](https://github.com/rweather/arduinolibs/blob/master/libraries/Crypto/AES128.cpp#L271). The hardware crypto does not help on tiny payloads (the ones we care about, [as demonstrated above](https://github.com/meshcore-dev/MeshCore/issues/259#issuecomment-4249586067)) and the header spells out that [none of this is hardened](https://github.com/rweather/arduinolibs/blob/master/libraries/Crypto/AESCommon.cpp#L37). If we ship a replacement AES implementation (that would make sense in any case if we remove the crypto lib), the LoC and complexity of reviewing it would also be increased.

erebion, 2026-09-15:
> This issue is stale because it has been open for 60 days with no activity. Remove the stale label or add a comment if this issue is still relevant, otherwise this issue will automatically close in 7 days.

Here's some activity to make sure security issues aren't simply forgotten.